Ultimaker/Cura #7069: Dark Theme colour palette not the same as macOS Dark Mode palette

kr15uk, 2020-02-05:
Cura 4.4.1 on macOS Catalina 10.15.2

Cura Dark Theme uses different colour palette than macOS inbuilt dark mode one.
That blueish hue feels very alien when flipping between other apps.

So I did quick find and replace in theme.json file.
39, 44, 48, 255 -> 41, 41, 41, 255
31, 36, 39, 255 -> 30, 30, 30, 255
75, 80, 83, 255 -> 30, 30, 30, 255
43, 48, 52, 255 -> 100, 100, 100, 255

Also that big blue banner while fine in light mode, in dark mode it's too "in your face" so I took this opportunity to add two amended lines from light mode theme.json
`"main_window_header_background": [41, 41, 41, 255],
"main_window_header_background_gradient": [100, 100, 100, 255],`

Welcome to use if you find this more fitting.

Ta.

Kris

Original
[image]

Amended
[image]

mahtDFR, 2020-02-10:
Thanks for providing this info. We had a quick discussion about this, the problem is that if we change the values here, they will not align with Ubuntu's system themes. It could be possible to set values per OS, but this will require more work. We won't do it for now unless there is significant demand.
Of course, we are always open to pull requests.

Ghostkeeper, 2020-02-10:
This is not deferred and we're not open for pull requests in this case, since Cura is Cura-themed not system-themed. We don't intend to align with your operating system's theme since it's impossible to make everything be correctly coloured together unless the entire interface is system-themed. And we'd still like to have the option to switch themes and define themes yourself.

closed 2020-02-10 by Ghostkeeper

nospam2k, 2021-05-17:
Why not just provide an additional mac-os-dark theme?

Ghostkeeper, 2021-06-01:
Not all MacOS dark themes are the same across the versions of MacOS we support. And there are a billion and one dark themes for Linux and Windows. We're not going to implement all of them. You're just going to have to get used to that Cura has its own style, I'm afraid.