Ultimaker/Cura #7071: Opening Marketplace dialog temporarily locks up Cura

fieldOfView, 2020-02-07:
**Application version**
4.4, 4.5, master

**Platform**
Windows 10

**Printer**
n/a

**Reproduction steps**
Open Marketplace
When the window opens, drag it around (smoothly)

**Actual results**
It takes a long time to open the dialog and populate it with the packages (15 seconds on my Core i7 system). Network traffic is only one or two seconds of that.
During the delay, the Cura window actually becomes unresponsive and the window will not follow the mouse cursor while dragging.

**Expected results**
The Marketplace opens in one or two seconds.

**Project file**
n/a

**Log file**
Nothing much to see

**Additional information**
Removing the ToolboxDetailList from ToolboxDetailPage.qml removes most of the delay, and I can't find out what I am missing without it.
Removing the `asynchronous: false` from ToolboxDetailList  also fixes this.

smartavionics, 2020-02-07:
Yes, it has been like this for a while. It has got noticeably worse in the last month or so.

fieldOfView, 2020-02-07:
\sa https://github.com/Ultimaker/Cura/commit/1487af167b616a190e320e6d05c94fad3a3ff95b

fieldOfView, 2020-02-07:
Profiling the non-responsiveness leads to `ListModel.setItems` for the models (or actually the `beginInsertRows` and `endInsertRows` methods). I'm thinking there's a signal storm involved.

valeria-gomez, 2020-02-09:
I am also on Windows 10. 
But Cura does not become unresponsive for me. 
The plugins just take a while to load on the first time only. 
When I close it and click on Marketplace again, it's instantaneously filled.

smartavionics, 2020-02-09:
Agreed, the first time the marketplace is opened, it takes ages. Subsequently, it opens quickly.

fieldOfView, 2020-02-09:
Please quantify “some time” and “ages”.

smartavionics, 2020-02-09:
Just tried it, first time it took approx 17 seconds to populate the dialog which was quicker than I thought it would be. Subsequent openings of the marketplace dialog take approx 1 second or less.

smartavionics, 2020-02-09:
The network part is relatively quick...

```
2020-02-09 21:39:01,414 - DEBUG - [MainThread] Toolbox.src.Toolbox._createDialog [216]: Marketplace: Creating dialog [Toolbox.qml].
2020-02-09 21:39:01,550 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [397]: file:///opt/cura/lib/cura/plugins/Toolbox/resources/qml/dialogs/ToolboxLicenseDialog.qml:42:9: QML Row: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2020-02-09 21:39:01,567 - DEBUG - [MainThread] Toolbox.src.Toolbox._makeRequestByType [535]: Requesting [packages] metadata from server.
2020-02-09 21:39:01,567 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [114]: No auth data to retrieve the access_token from
2020-02-09 21:39:01,568 - WARNING - [MainThread] Toolbox.src.UltimakerCloudScope.request_hook [22]: Cannot add authorization to Cloud Api request
2020-02-09 21:39:01,568 - DEBUG - [MainThread] Toolbox.src.Toolbox._makeRequestByType [535]: Requesting [authors] metadata from server.
2020-02-09 21:39:01,568 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [114]: No auth data to retrieve the access_token from
2020-02-09 21:39:01,569 - WARNING - [MainThread] Toolbox.src.UltimakerCloudScope.request_hook [22]: Cannot add authorization to Cloud Api request
2020-02-09 21:39:01,569 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: FastView
2020-02-09 21:39:01,570 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: GodMode
2020-02-09 21:39:01,570 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: RawMouse
2020-02-09 21:39:01,570 - DEBUG - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [270]: Found a plugin that was installed with the old plugin browser: YappiProfiler
2020-02-09 21:39:02,324 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [051af5586e484a6880ae07efd98daf78] started
2020-02-09 21:39:02,325 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [040d1e156e4a4ed0b1e7d0d61534f411] started
2020-02-09 21:39:02,326 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-02-09 21:39:02,981 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [040d1e156e4a4ed0b1e7d0d61534f411] finished.
2020-02-09 21:39:02,982 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-02-09 21:39:06,516 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [051af5586e484a6880ae07efd98daf78] finished.
2020-02-09 21:39:06,517 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop

```

nallath, 2020-02-10:
The marketplace code is pretty bad. It's mostly the way that the QML is built up that makes it really slow. 

I've tried making it asynchronous, but then other bugs pop up. I hope to be able to refactor a bunch of the code for the 4.6 release.

ChrisTerBeke, 2020-02-10:
@nallath maybe that would be a good opportunity to make use of the pagination options on the API as well? That should speed up initial data load considerably.

Ghostkeeper, 2020-02-11:
That'll reduce the load on the networking part, but most of it is in the QML loading for some reason. And I don't know whether the QML loading is really going to speed up, considering it still needs to filter there and then request an arbitrary number of pages.

This QML code and the Python models belonging to it is due a refactor. I think the terror factor of that dragon has gradually been growing recently.

fieldOfView, 2021-06-11:
Terror-factor of redoing the code notwithstanding, I think the recent issue where upgrading to Cura 4.9 caused a crash with an old version of a plugin with an update of that plugin had been available for a month shows that it is imperative that the Marketplace is improved. In the current state, where Cura is locked up for a good minute on my system, you cannot blame users for not using the Marketplace to keep their plugins (and materials) up-to-date.

nallath, 2021-06-14:
I totally agree. But as I've already mentioned in another topic; the refactor of the marketplace is scheduled to take place very soon. 

The current setup isn't only hard on the local machines, it's also wreaking havoc on our servers (since it always requests _all_ plugins instead of using pagination)

fieldOfView, 2021-06-14:
>  it's also wreaking havoc on our servers (since it always requests all plugins instead of using pagination)

Sounds like something that could be easily solved with server-side caching, but I'll not get into that again.

fieldOfView, 2022-04-21:
I'll close this with the imminent replacement of the Marketplace dialog with a new one in the 5.0 beta.

closed 2022-04-21 by fieldOfView