Ultimaker/Cura #7072: Default CR-10 start G-code has extra Z step

vakker, 2020-02-07:
**Application version**
4.4.1

**Platform**
Ubuntu 18.04

**Printer**
Creality CR-10

**Reproduction steps**
Just install and choose CR-10 as the printer.

I have in the Start G-code section:
```
....
G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up <-------------------------------- first up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up <-------------------------------- second up
```

This can not print, as after drawing the first 2 lines it moves up 1 layer and then starts the actual print.

**Expected results**
Commenting the last line out does provide a good setting, so I was able to print without problems.

Maybe I'm missing something, but the default should be working without issues.

Ghostkeeper, 2020-02-11:
I think that's correct actually. You need to look a bit further than just the start g-code though.

```
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up  <----------------------------move up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position  <-------move diagonally down to start
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up  <----------------------------move up

G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:197
;LAYER:-5
M107
G0 F9000 X129.927 Y137.436 Z0.96 <--------------------------move diagonally down to start
;TYPE:SUPPORT-INTERFACE
G1 F2700 E0
G1 F1125 X130.598 Y136.833 E0.28805 <-----------------------printing brim
G1 X130.714 Y136.737 E0.33613
```

It's making those diagonal moves since otherwise it would move just 0.28mm over the build plate. If the build plate is not entirely level, that could damage the build plate from the nozzle scraping over the plate.

vakker, 2020-02-12:
But I'm not printing brim.
For me the first layer just goes in the air without touching the surface.
After commenting the second `move up` statement, it does work for me with a properly levelled build plate.

Ghostkeeper, 2020-02-13:
If you're not printing a brim, it would just start printing whatever else is first, being a skirt, raft, support, prime tower, prime blob or just the model itself. The comment was meant as an illustration of where it was starting the actual print.

Can you copy a bit more of your g-code, up until the point where it actually starts printing something? So far I'm not able to reproduce your problem.

vakker, 2020-02-17:
So my gcode is:
```
;FLAVOR:Marlin
;TIME:24555
;Filament used: 17.914m
;Layer height: 0.2
;MINX:103.489
;MINY:112.651
;MINZ:0.2
;MAXX:188.241
;MAXY:194.261
;MAXZ:90
;Generated with Cura_SteamEngine 4.4.1
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
;G1 Z2.0 F3000 ;Move Z Axis up
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:450
;LAYER:0
M107
G0 F6000 X116.643 Y123.451 Z0.2
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X117.028 Y123.101 E0.01731
G1 X117.967 Y122.287 E0.05864
G1 X118.22 Y122.075 E0.06962
G1 X119.017 Y121.425 E0.10382
G1 X119.402 Y121.124 E0.12008
G1 X120.297 Y120.458 E0.15718
G1 X120.573 Y120.259 E0.1685
...
```

Ghostkeeper, 2020-02-25:
So your g-code contains these lines just before printing the skirt:

```
;LAYER:0
M107
G0 F6000 X116.643 Y123.451 Z0.2
;TYPE:SKIRT
```

That means that it's instructing the print head to go to `Z0.2`. This makes sense if your initial layer height is 0.2mm. Is it not going to `Z0.2` then?

vakker, 2020-02-28:
I don't think so, or there's something else going wrong.
Currently I'm printing with that line commented out and it seems to work well. However when I have that line in, all the prints are messed up. The first layer starts higher than is should and it doesn't stick to the plate.

Line commented out:
[image]

Default setup:
[image]

Ghostkeeper, 2020-03-04:
I think you'll want to reduce your Initial Layer Height setting then, or increase the Initial Layer Line Width.

That travel move there is necessary to print with the correct Initial Layer Height. If you remove it, you will get overextrusion in your first layer (because it's too close to the bed to extrude the amount that it wants) and then underextrusion in your second layer (because it's too far from the lowered first layer). Furthermore your first skin line will start from the wrong location and then be practically missing, since it's not extruding enough for the line that suddenly became 10x as long.

vakker, 2020-03-04:
I removed it and I'm not getting overextrusion in my first layer and then underextrusion in my second layer. However, if I leave it in then I get underextrusion in my first layer.

I don't know what's the real cause of the issue, so I might be wrong with accusing this specific line, but something is definitely off with the default settings.

Also, as you pointed out `G0 F6000 X116.643 Y123.451 Z0.2` should just tell the extruder to go to Z=0.2mm so I don't understand how it has an impact on the initial layer at all.

Ghostkeeper, 2020-03-09:
We can't really optimise your profiles for you from here. Only if we can understand why a change is needed will we be able to make the change. If we remove the movement to the initial layer's Z coordinate this is going to wreak havoc with almost every print out there.

vakker, 2020-03-09:
It's not really optimising for me, I managed to reproduce the same behaviour on 2 different CR-10 machines.

If you go to a gcode simulator, e.g. http://nraynaud.github.io/webgcode/ and simulate the following with and without the line marked (if you rotate it to the front view you'll see the issue): 
```
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line

G1 Z2.0 F3000 ;Move Z Axis up <- this

G1 F1500 E-6.5
;LAYER_COUNT:450
;LAYER:0
M107
G0 F6000 X116.643 Y123.451 Z0.2
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X117.028 Y123.101 E0.01731
G1 X117.967 Y122.287 E0.05864
G1 X118.22 Y122.075 E0.06962
G1 X119.017 Y121.425 E0.10382
G1 X119.402 Y121.124 E0.12008
G1 X120.297 Y120.458 E0.15718
G1 X120.573 Y120.259 E0.1685
```
then you'll see that everything after that line is printed at Z=2.0mm. I don't see how that can be correct for anyone. However, if it's removed then it prints the first layer at the same height as the initial 2 lines. If the first 2 lines are printed at Z=0.28, then why would the first layer be printed at Z=2.0?

Also, as you stated at the beginning:
```
G0 F9000 X129.927 Y137.436 Z0.24 <--------------------------move diagonally down to start
```
this diagonal down movement is not happening.

However, if I break up that line to:
```
G0 F6000 X116.643 Y123.451
G0 Z0.2
```
then it does do that movement.
Of course the simulator could be wrong, but I see the same behaviour on the printer.

Ghostkeeper, 2020-03-13:
On my screen by just copy-pasting your g-code, that simulator does display a diagonal move downwards to the build plate:

[image]

The diagonal movement clearly specifies Z0.24, so that is what the printer must execute. If it's not doing that, the printer is really at fault.

vakker, 2020-03-23:
Yes, you are right, I was looking at the simulation the wrong way.
I still don't know why it causes issues on the printer. It might be a backlash issue, so when it goes up, it doesn't go back down to the right level. I'll investigate further and reopen the issue if necessary.
Thanks!

closed 2020-03-23 by vakker