Ultimaker/Cura #7073: Ability to set persistent printer profile for different computer users

3DEnthusiast, 2020-02-07:
When someone logs onto a computer, it creates a computer profile specific to that person. Each person to log onto a single computer will have a different profile on the same computer. From that idea, each new or different person to log onto a single computer has to specify the printer profile in Cura the first time they run the program. 
 
I would like to know if setting a single, default printer profile is possible. That way, if someone new logs onto the computer and opens Cura it will have the default profile set without having to go through the Printer Setup Wizard.

valeria-gomez, 2020-02-09:
I like the idea, but I don't think Cura is intended to work that way.

That being said, they have Cura Backups that keeps and restores your whole configuration. I personally use it quite a bit and went for the full feature that works together with Ultimaker Accounts. In the earlier Cura builds, I kept losing my previous version configuration. My thanks to whoever created Cura Backups feature. 

When you restore backups, it will get it from the logged-in account.

3DEnthusiast, 2020-02-10:
I'm trying to set Cura up on a student used computer so that multiple students in different class periods can use it without each person having to run the setup.

Ghostkeeper, 2020-02-11:
Maybe an easier way to do this is to save a project file with your printer, using File -> Save... and then share that project file in a central place where all users can access it.

For Ultimaker's printers this is solved by discovering printers on the network. It's very easy to set up the correct printer that way. Alternatively the printers can be assigned to people's Ultimaker accounts in a team so that they can print via the cloud. But this only works for Ultimaker printers out of the box. You could set up something similar using OctoPrint if you have a different printer.

I've heard that some people like to set up a symlink from the Cura configuration directory to redirect to Dropbox. That way they could sync their Cura directories via dropbox. You could do something similar by just symlinking it to a local folder that all users can access.

I don't think we'll implement this particular feature request because of the difficulty in separating the user accounts. But there is something to be said for configuring the configuration directories as administrator. It's not simple because of the differences in OS. Currently Cura follows the OS conventions (AppData/Local and Roaming for Windows, Application Support for MacOS, .config and .local and .cache for Linux). And there's also the question of where it store the preference of where the preferences should be stored.