Ultimaker/Cura #7074: [4.4.1] Linear-advance-like effect?

Maeyanie, 2020-02-08:
**Application version**
4.4.1

**Platform**
Win10 x64, GTX 1070

**Printer**
Custom FFF

**Reproduction steps**
1. Enabled "Enable Bridge Settings"
2. Slice model
3. Look at bridging layer

**Screenshot(s)**
[image]

**Actual results**
Cura seems to produce artefacts when speeding up after a slow bridge, similar to what linear advance would do. However, I don't see any way to configure or disable this, and it is noticeable as a blob in the print.
When I set the bridge speed to the same as my print speed the blob no longer shows in the preview, however bridging at high speed has its own problems, as would reducing the print speed to match the bridge speed.

**Expected results**
The printed lines should be even and smooth. Or at least there should be a way to configure/disable this feature.

**Project file**
[base.plain.square.inch.2x1.openlock.triplex.3mf.zip](https://github.com/Ultimaker/Cura/files/4173827/base.plain.square.inch.2x1.openlock.triplex.3mf.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4173824/cura.log)

**Additional information**
Changing the settings of "Equalize Filament Flow" and acceleration don't seem to matter, 
There are a rather lot of settings in Cura, however, so it's distinctly possible there is a way to disable this and I'm just missing it.

smartavionics, 2020-02-08:
What it does is try to compensate for the under-extrusion that occurs after a bridge wall line is printed. That happens when the speed and/or flow for the bridge line segment is less than the speed and/or flow for the following non-bridge line segment. It looks a lot more pronounced on the layer view than it does on the printed object.

Maeyanie, 2020-02-08:
This may be useful on a printer which does not itself support linear advance, but on mine (which does), the double-compensation is noticeable. If I disabled that on the printer, then the quality of the corners (which Cura doesn't compensate in this manner) would suffer instead.

I certainly wouldn't remove this feature, but having the compensation configurable would be an improvement even on Ultimaker printers since different filaments compress differently and so need different compensation.

smartavionics, 2020-02-08:
Yes, I think it would be good to be able to specify the amount of compensation, I will do that.

smartavionics, 2020-02-08:
BTW, I'm not convinced that linear advance is good for bridging as it means that towards the end of the bridge lines, the extrusion rate will alter.

Maeyanie, 2020-02-08:
If it's configured correctly, the extrusion rate should change proportionally to the speed change, so that the flow of filament is constant.
This does need careful calibration though. If it's over- or under-set, you're right, it can very easily cause bridges to fail.

smartavionics, 2020-02-08:
My next Cura build will have a Bridge Wall End Boost setting that controls this behaviour. I will also submit a PR to incorporate that into UM's releases in the future.

You can get my Cura builds from https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there for more information. I expect I will make a new build later today or tomorrow.

Maeyanie, 2020-02-08:
Much appreciated, thanks. I'll give that a try.

smartavionics, 2020-02-08:
There's now a 20200208 build that has this new setting. It defaults to 100% so as to keep the existing behaviour, set it lower (or even 0) to reduce the effect.