Ultimaker/Cura #7075: Estimated print time is out by almost 100% (actual time is about double)

NameOfTheDragon, 2020-02-08:
**Application version**
(The version of the application this issue occurs with.)
4.4.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Pro 1909

**Printer**
(Which printer was selected in Cura?)
Creality Ender 3

**Reproduction steps**
1. Download model from Thingiverse (https://www.thingiverse.com/thing:3502973)
2. Load the `Spool_Holder_V1.stl` into Cura
3. Select 0.16mm layer height, 100mm/s print speed. Material is PLA, 1.75mm filament
4. Slice and send to Octoprint.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
1. Image showing Windows 10 version, Cura version and Cura estimated print time 3h 36m [image]
2. Image showing Octoprint print in progress, with estimated total print time, time so far and job % complete. 
[image]

Note: the Octoprint statistics are internally consistent. 14% of the job took just over 51 minutes to complete, so 100% of the job would project to about 6 hours, which is consistent with Octoprint's estimate of 5h 50m.

**Actual results**
(What happens after the above steps have been followed.)
Cura predicts a print time of 3h 36m which is a gross error.
Octoprint predicts 5h 50m, not far off double the Cura estimate.
Octoprint's estimate is "about right" based on 15% job progress.

**Expected results**
(What should happen after the above steps have been followed.)
Cura should make a more realistic print time estimate. Doubling the time would get a lot closer than the current estimate, which must be considered a gross error. An error of +/-10% would be more reasonable.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
I used this file to test: https://www.thingiverse.com/download:6230500

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
Not included as probably not relevant to the print time estimate.

**Additional information**
(Extra information relevant to the issue.)

valeria-gomez, 2020-02-09:
Is this because your printer is not Ultimaker? 

A few years back, a cura developer answered a similar question. From what I can recall, Ultimaker's printer settings have more accurate values (jerk, acceleration). Cura uses those to estimate the print time. I have an Ultimaker 3 and my estimated and actual print times are rarely misaligned. When they are, the difference is negligible. 

In this case, perhaps the printer settings of your printer have to adjust. In turn, Cura will be able to give a better estimation.

Perhaps a developer can shed more light on this, but this is what I remember.

nallath, 2020-02-10:
> Perhaps a developer can shed more light on this, but this is what I remember.

You're completely right! We (Ultimaker employees) didn't create the profiles for third party printers and we also don't test them (as this would take too much time and we would have to buy all those printers). As such, the non-Ultimaker printer profiles will have varying quality.

closed 2020-02-10 by nallath