Ultimaker/Cura #7077: Trying to print something, and can't seem to change support pattern.

CptanPanic, 2020-02-08:
**Application version**
4.4.1

**Platform**
Win64

**Printer**
Ender 3

**Reproduction steps**
1. Added support
1. Changed support pattern to cocentric, view changes.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
Support pattern doesn't seem to have any affect.

**Expected results**
Should see something like zig-zag or cocentric.

valeria-gomez, 2020-02-09:
Is it the Support Interface Pattern that you want to change? The baby blues in the middle look like the SIPs. The regular support has cyan color. 

[image]

rburema, 2020-02-11:
Closing this as @valeria-gomez is right. If you still experience (related) problems, feel free to reopen. (Glad we made the difference between normal and SIP more clear a few versions ago...)

closed 2020-02-11 by rburema