Ultimaker/Cura #7079: Dual Extruder - Different Build Plate Adhesion setting

BR365, 2020-02-09:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Geeetech A20M - Marlin

**Reproduction steps**
1. Import of two STL files
2. STL1 is assigned to left extruder
3. STL2 is assigned to right extruder
4. Activate Build Plate Adhesion for left extruder
5. Deactivate Build Plate Adhesion for roght extruder

=> Build Plate Adhesion for left extruder is deactivated, too!

It is not possible to activate Build Plate Adhesion only for ine of the two extruders.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
It is not possible to activate Build Plate Adhesion only for ine of the two extruders.

But this can be necessary e.g. when printing a surface in black with text in red. The surface should be printed with brim, but the text shouldn't be printed with brim.

**Expected results**
Build Plate Adhesion settings for left and right extruder should be independent from each other.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
[image]
[G_Gleisbox-Deckel-mit-Logo-Ausschnitt.zip](https://github.com/Ultimaker/Cura/files/4177058/G_Gleisbox-Deckel-mit-Logo-Ausschnitt.zip)

**Log file**

**Additional information**

Thanks,
Moritz

Liger0, 2020-02-09:
#6996

Ghostkeeper, 2020-02-12:
In your example, I wouldn't expect the Märklin text to get a brim since the brim is still obstructed by the surrounding square. Something is weird with the model though. It seems to think that the two models are overlapping. This way, Alternate Mesh Removal is kicking in, removing half of the layers for one model and half of the layers of the other one, resulting in this weird brim around just the text. The first layer of the square is actually floating.
[image]

Your feature request - to make the Adhesion Type setting a per-object setting - is not implemented yet for two reasons:
* If a brim collides with another object that doesn't have a brim, it needs to interrupt the line. This defeats the purpose of the brim to get a consistent flow, but it's possible to make this work. It would currently require a refactor on our end that we haven't found the time for.
* The raft adds layers below the print. These layers are added globally; we don't track layers per object or per extruders. To fix this, we'd need to throw the implementation around to move the model up. This requires removing the feature where the raft air gap can be any size rather than being constrained to a multiple of the layer thickness. It'll also make support generation a bit more complex since the support needs to be generated only up until the raft.