Ultimaker/Cura #7080: Materials not showing up in CURA

3dprosthesis, 2020-02-10:
cura 4.4.1
Mac High Sierra 10.13.6
Creality cr10s Pro
Downloaded From Marketplace the "Matterhackers Nylonx" material profile.
It does not show up in the material selection menu.

I have seen this issue spoken about but no cure. There was one suggestion that if the filament thickness is different from the printers therefore that is the cause of it not showing up. There is no mention though on the download page on what filament thickness the profile. 

My question is, is there an easy way to get settings for the nylonx to function? I have not much experience with cura so any step by step precise instructions would be enormously helpful

Warmest
Christina
NYC

mahtDFR, 2020-02-10:
Some things to try:
Probably stating the obvious: Cura needs a restart before the material registers.
Other than that it could be an incompatible nozzle size.

If you share a zipped 3mf I can see your settings and try to help. A log would be useful too.

fieldOfView, 2020-02-10:
Unless I am mistaken, the profiles one can download from the Marketplace are only 2.85 mm profiles, which will not work on any Creality machine because those are 1.75 mm machines.

nallath, 2020-02-10:
> Unless I am mistaken, the profiles one can download from the Marketplace are only 2.85 mm profiles, which will not work on any Creality machine because those are 1.75 mm machines.

You're right. You can only download 2.85 materials on the marketplace.

closed 2020-02-10 by nallath

3dprosthesis, 2020-02-10:
is there any way to copy the settings from the 2.85 mm material settings and paste them or manually enter them into a custom profile for 1.75mm

fieldOfView, 2020-02-10:
What you download from the Marketplace are profiles tuned for specific (Ultimaker) printers. There's no guarantee these materials will print ok with other printers, let alone that the same settings will work.

Ghostkeeper, 2020-02-12:
Yeah, you can modify these profiles. To do that, add a printer that has the correct diameter (such as Ultimaker S5), then go into the Material Manager. It'll list that material there. Click on it and adjust the diameter to be correct for the filament you bought.

No guarantees on print quality of course, but the printing temperature and cooling settings there should be reasonably okay.