Ultimaker/Cura #7081: [4.4.1] Hammering the Serial Port -- again

FlailAway, 2020-02-10:
**Application version**
4.4.1

**Platform**
Mint 19.3 + Cinnamon, all u to date

**Printer**
Custom

**Reproduction steps**
Start Cura -- I thought this issue was fixed, but I skipped 4.3.x but just went back to it and it too is doing it.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Our Arduino IoT Serial port gets closed. There are four options, 
"ttyACM0" 
"ttyACM1" 
"ttyUSB0" 
"ttyUSB1" 
and all four get closed.

**Expected results**
Leave the Serial ports alone

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4182265/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/4182267/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/4182268/stdout.log)

**Additional information**
Not sure if it is any help but I have disable a lot of stuff in Marketplace to try and fix this and .log mentions them as disabled. But then, later on in several places it says, "updating ..." and/or "looking for older ..."

It seems to happen about a minute or more after starting Cura and/or after saving a file, but that's just a guess as I cannot tell exactly when it hammers the Serial port. But it **only** happens with Cura and it happens **every** time I run Cura.

On another issue, why do you have two places to store stuff? ".config" and ".local?" Why not in just one or the other?

fieldOfView, 2020-02-10:
If you don’t need USB printing support in Cura, you can disable that plugin via the Installed tab of the Marketplace.

FlailAway, 2020-02-11:
> If you don’t need USB printing support in Cura, you can disable that plugin via the Installed tab of the Marketplace.

Thanks, but I am not the village idiot and I suggest you re-read the OP **and** look at the logs to see what has been disabled in the marketplace.

I have had this issue before and it was corrected without comment. But it is appearing again so I suspect some earlier code has been re-used and brought it back into play in 4.4.1.

fieldOfView, 2020-02-11:
> I am not the village idiot and I suggest you ...

Yeah, no, sorry, that must be me, for spending my time to try and help you. Bye!

Ghostkeeper, 2020-02-12:
> On another issue, why do you have two places to store stuff? ".config" and ".local?" Why not in just one or the other?

This is to comply to the XDG base directory specifications. See also [this reference](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html). We also use `~/.cache` by the way.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/5207, because that's what it is. Apparently the workaround suggested workaround doesn't work for you any more, but let's discuss that in the appropriate issue.

closed 2020-02-12 by Ghostkeeper