Ultimaker/Cura #7082: [4.4.1] Selecting Settings in Mesh Type dialog changes mesh type back to Normal Model

pfjason, 2020-02-10:
**Application version**
4.4.x

**Platform**
Ubuntu 19.10

**Printer**
Any

**Reproduction steps**
1. Add a model to the build plate
2. Add a support blocker block to the model and select it
3. Change the mesh type to "Modify Settings for Overlaps"
4. Click "Select Settings" and select any setting.
5. Notice that the Mesh Type is now "Normal model"

**Screenshot(s)**
[image]

[image]

[image]

**Actual results**
Mesh Type for the support blocker / Settings Modifier block is now "Normal Model" and slicing the model will now produce a cube in that spot. 

**Expected results**
Mesh Type should not be changed

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

valeria-gomez, 2020-02-11:
This is a bug in 4.4.1. 

I'm literally checking out Cura 4.5 at this moment.  It looks like it was fixed in the recent 4.5. Huurah!

https://github.com/Ultimaker/Cura/releases/tag/v4.5-beta

Ghostkeeper, 2020-02-12:
Yup, we found this too and fixed it. Thanks for the report nonetheless!

closed 2020-02-12 by Ghostkeeper