Ultimaker/Cura #7083: Can't display model at Preview mode

joonghochoe, 2020-02-11:
**Application version**
Cura 4.4.1, 4.3.0, Master

**Platform**
Windows 10 1909, Intel HD Graphics 630

**Printer**
Custom FFF, HMS434

**Reproduction steps**
1. Install Cura
2. Chose Custom FFF or HMS434
3. Setting 8 for Number of Extruders (X : 300, Y : 300, Z : 300)
4. Load Model
5. Multiply model to 2
6. Assign 2nd model to Extruder 8
7. Slice
8. Change to Preview mode

**Screenshot(s)**
[image]

**Actual results**
The second model I assigned to extruder 8 is not displayed. 

**Expected results**
I want show all model at Preview mode.

**Project file**
[CFFFP_cylinder.zip](https://github.com/Ultimaker/Cura/files/4183708/CFFFP_cylinder.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/4183706/cura.zip)

**Additional information**
On some PCs the above problem appears. On other PCs, there is other problems at preview mode.
1. Setting 8 to Number of Extruders
2. Load Model and Multiply 7
3. Assign each 8 extruder to 8 models
4. Slice
5. Change Preview mode
6. Uncheck Extruder 1 or 4 at Color scheme
7. Disappear Model 1(or 4) and Model 5, 6, 7, 8 at the same time.

rburema, 2020-02-11:
Confirmed, even just setting a single model to 8 will do it (in my case at least), no need for the other reproduction steps. Looks like an 'off by one' bug in a comparison somewhere.

Ghostkeeper, 2020-02-12:
I remember this being brought up with the team, but what was our conclusion? We try this as a small fix?

rburema, 2020-02-12:
Yes, we'll try to timebox it. I've already spent some time of the timebox, but it wasn't immediately obvious.

joonghochoe, 2020-05-02:
Any good news about this bug? This bug still seems to remain in version 4.6.

Some desktop PC(Windows 10 Pro 64-bit (10.0, Build 18363), Intel(R) HD Graphics 630)
1. Solid View is OK
[image]

2. Layer View, Simulation View is not OK, Only at 8th model
[image]
[image]

Another desktop PC(Windows 10 Pro 64-bit (10.0, Build 18363), Intel(R) HD Graphics 4600)
1. Extruder 1 is checked at Layer View
[image]

2. Uncheck only extruder 1
[image]

rburema, 2020-05-06:
I _think_ I fixed this with these PR's:
 https://github.com/Ultimaker/Uranium/pull/606
and
 https://github.com/Ultimaker/Cura/pull/7693

_If_ so, it'll probably be fixed in 4.7

closed 2024-11-24 by joonghochoe