Ultimaker/Cura #7084: Feedrate preview colours don't match expected results.

Lieste, 2020-02-11:
Cura 4.4.1 

Win10Home x64 GTX970

Ender 5 Plus

Setting a thin outer wall, with a thicker inner wall. 0.4mm vs 0.6mm, and selecting a speed which should give approximately the same feedrate 90mm/s vs 60mm/s results in different feedrates being shown in the preview. The outer wall being faster than the inner.
To obtain the same nominal feedrate, the advance speeds for both need to be the same (e.g. 60mm/s) despite the difference in width. I also see difference in the skin additional walls vs skin, and have to set a non consistent width/speed set to keep these and supports reading the same feedrate colours.

With 90mm/s and 0.4mm outer walls, and 60mm/s 0.6mm inner walls, the two lines have significantly different colours.
**Expected results**
These should have approximately the same feedrate - by intent.

smartavionics, 2020-02-11:
I think the problem is that the "feedrate" is actually the print speed, i.e. speed of nozzle movement, not the rate that the extruder is pushing plastic. Arguably, calling it "feedrate" is incorrect or at least misleading.

Ghostkeeper, 2020-02-11:
Yes, the term "feedrate" derives from CNC milling and denotes the surface movement speed, not the extrusion rate.

The term comes from how much material the teeth of the CNC bit are carving away with each rotation. Cura uses that term by convention.

Lieste, 2020-02-11:
The calibration seems to suggest the extruder advance rate though - on the order of 3-4mm/s for a 60-90mm/s speed of print. It is neither one thing nor the other.

Ghostkeeper, 2020-02-12:
We've renamed "feedrate" to "speed" now in layer view. We discussed it briefly and agree that this is better: https://github.com/Ultimaker/Cura/commit/b0418716fd999010f30f0d60e292b8c653072c77

The minimum speed on the lower end of the literal spectrum is probably actually in the print. Even if none of the speed settings are listing that speed, it could still reach that low a speed because of the minimum layer time. It'll slow down the nozzle in order to make the layer take longer to print, so that it can cool down and solidify before the next layer is applied. If the legend contains speeds that are not actually in the print, please open up a new issue for it. Because the legend should only display speeds that are in the print!

closed 2020-02-12 by Ghostkeeper