Ultimaker/Cura #7085: how to make the tool change like this

lyfbuaa, 2020-02-11:
Application version
Ultimaker_Cura-4.2.1.AppImage

Platform
fedora 31 x86_64

Printer
diy printer like ultimaker with marlin 1.1.9

object:
two extruder: T0 T1,T1 when working is lower than T0.Therefore  when T1 is working ，Z is lower than when T0 is working.    How to execute the Z movement?     like the dual extruder ultimaker?

rburema, 2020-02-11:
Do you mean physically? Dual extrusion Ultimaker printers do that on tool-change. That is to say, if a T1 command is given, T1 is set to the 'enabled' (for the moment) position and T0 to the 'disabled' (for the moment) position. It's not something we otherwise do in Cura (at least not for UM printers).

I assume other printers are similar. Do you already have your printer set to multiple extruders? You can change the number of extruders in the machine settings menu.

lyfbuaa, 2020-02-11:
I know set  the number of extruders ,but don't know  how to execute the Z movement like that the different height nozzle of  the dual extruder ultimaker is working.

mahtDFR, 2020-02-12:
With Ultimaker 3 printers and above, the lift switch mechanism is mechanical, operated by a switch in the print head. To switch heads, it will mate with a switch in the corner of the gantry. Therefore, the 'T' commands _are_ the thing that makes the height difference work.

If your nozzles are locked together, this cannot be solved with gcode without a mechanical solution. Can you give some details about the printer you're using?

Ghostkeeper, 2020-02-12:
Ultimaker printers calibrate the difference between the nozzle heights at the same time as their bed levelling procedure. And then they offset the print head in the firmware. It's not left up to Cura to compensate for the offset.

If your g-code is in relative coordinates rather than absolute coordinates, a search-and-replace post-processing script can implement this trivially (replace `T0` with `T0\nG0 Z-0.7` or something). If it's absolute, I think you'll need some Python experience to write your custom post-processing script to implement this.

lyfbuaa, 2020-02-13:
> With Ultimaker 3 printers and above, the lift switch mechanism is mechanical, operated by a switch in the print head. To switch heads, it will mate with a switch in the corner of the gantry. [...]

I am diying my dual extruder printer.   I check the dual extruder design of um3.  I think i can make the dual extruder mechanically that has the switch to control up and down of T1。 But I still don't know how to contol the Z  up and down prior to T1 in printing.     
for example:  I  add "G1 Z1" in T1 start code.   
Actual results:  after execute Z1,  the Z reset before T1 print.  
Expected results:after execute Z1,T1 print.  after T1 print,Z reset or Z-1.Then T0 print.

lyfbuaa, 2020-02-13:
> Ultimaker printers calibrate the difference between the nozzle heights at the same time as their bed levelling procedure. And then they offset the print head in the firmware. It's not left up to [...]

The offset effect the T0 and the T1 at the same time. But I only want it to effect the T1.
Expected results:after execute Z1,T1 print. after T1 print,Z reset or Z-1.Then T0 print.
I think ?:
In T1 start code:
M401 
G1 Z1
G91
In T1 end code:
G1 Z-1
G90

Ghostkeeper, 2020-02-13:
That would effectively run everything for `T1` in relative g-code but everything for `T0` in absolute g-code. Try this instead, to just do the one command in relative mode:

T1 start g-code:
```
M401 # Don't know why you need this but I'll keep it in?
G91
G1 Z1
G90
```

T1 end g-code:
```
G91
G1 Z-1
G90
```

lyfbuaa, 2020-02-13:
G91 should before and I make a mistake。But the code can't get the expected results.
Actual results: after execute Z1(G91),the Z reset before T1 print.

the slice code in attachment.In order to easy to watch,I write Z10 instead Z1.

 :

Ghostkeeper, 2020-02-14:
How about this then:

Enable the "Relative Extrusion" setting.

Then set the T1 start g-code to:
```
M401
G1 Z10
```

And the T1 end g-code to:
```
G1 Z-10
```

lyfbuaa, 2020-02-15:
It seems the same thing.I can't watch if xye is relative,but z shouldn‘t . 
I think whether that g92 can set z position that M401 records can solve it or not?
But I don't know how to pass the value in gcode.

lyfbuaa, 2020-02-16:
now I can solve it in marlin,but I still want know whether or not it can be solved in cura .

Ghostkeeper, 2020-02-17:
Sorry, I made a mistake. Relative Extrusion of course only applies to the E axis.

I think you'll have to resort to implement this as a post-processing script then, that basically just offsets all Z coordinates for the other extruder.
I wonder why the `G91` and `G90` trick doesn't work though.

lyfbuaa, 2020-02-19:
If modify T1' start code as below ,Z keep the target position. But xy movement is abnormal.

G91
G1 Z1
;G90

Block137, 2020-02-24:
You need to put G90 to revert back to absolute mode because X,Y is always in absolute mode

G91
G0 Z1
G90
...

This will be messed up after a layer change anyway. This cant be done in Cura, it has to be done in firmware.

lyfbuaa, 2020-02-24:
oh,I see

Ghostkeeper, 2020-02-25:
Aside from helping you set up an extruder switch code, we're not going to implement anything in Cura to really help you with this out of the box. This is the domain of firmware and the custom extruder start/end g-codes. You'll have to stick with your firmware setting adjustments. Sorry!

closed 2020-02-25 by Ghostkeeper