Ultimaker/Cura #7090: [4.4.1] Can't "Pause at height" with Ultimaker S5

Bastian-Reul, 2020-02-11:
**Application version**
Cura 4.4.1

**Platform**
Windows 10 Pro N 64 Bit

**Printer**
Ultimaker S5
Firmware: 5.4.27.20191212

**Reproduction steps**
1. Load an STL File to Cura
2. Extensions-> Post Processing -> Modify G-Code
3. "Pause at height"
[image]

4. Slice
5. Print over Network
 -> Print is done without Pausing
6. Save the File als G-Code, no M0 Statement found in G-Code
[image]

**Actual results**
Ultimaker S5 didn't Pause at my heigth. Neither via height in cm or in Layer Number

**Project file**
[UMS5_StopAt-Z.zip](https://github.com/Ultimaker/Cura/files/4184997/UMS5_StopAt-Z.zip)

Searching for help,
sincerly, Bastian

valeria-gomez, 2020-02-11:
Hi there, we also use version 4.4.1. (now moved to 4.5) 

Can you try removing that pp plugin and adding it again? Slice and save gcode. 

I know this works because our new operator today forgot that this was saved in all of the gcode files. We were confused because all prints were pausing at the same layer height. He was using Cura version 4.4.1. I can confirm it was working for us - by accident.

Bastian-Reul, 2020-02-12:
Hey Valeria,
thanks a lot for the hint, but it didn't changed anything. Any other suggestions?
Sincerly, Bastian

mahtDFR, 2020-02-12:
> Hey Valeria,
thanks a lot for the hint, but it didn't changed anything. Any other suggestions?
Sincerly, Bastian

Can you please check in 4.5 (beta) and report back? There was a commit that should fix this.
https://github.com/Ultimaker/Cura/pull/6690/commits

Bastian-Reul, 2020-02-13:
Hey mahtDFR,
it seems that the update fixed the problem:

[image]

I'm printig actually a 30h print, but on Monday I will give it a try!
Thanks a lot for so long!

Sincerly, Bastian

mahtDFR, 2020-02-13:
> Hey mahtDFR,
> it seems that the update fixed the problem:

Good to hear. Good luck with the 30h print.

closed 2020-02-13 by mahtDFR