Ultimaker/Cura #7092: Cura 4.4.1 - Pause At Height - Wrong/Missing Z Position

Kayu84, 2020-02-11:
**Application version**
4.4.1

**Platform**
Windows 10 Pro 64bit
i5 3,5GHz
32GB Ram

**Printer**
Anycubic i3 Mega

**Reproduction steps**
I Have 6 stl files, produced with onshape.
They all have the same height, same size
differce in shape & holes

Placing them 1 by 1 into cura,
adding "post processing script": pause at height for repetier
slice and save into their own gcodes

Pause repetier:
```
height 0.6
X 0
Y 0
Z 20
Ret 5
Ex 20
Re 0
```

**Screenshot(s)**
[image]

**Actual results**
The Screenshot shows the issue
All files showing the extruder moving on layerheight z=0.6 before pause
the "current z" state differece
and after the pause there is a missing: G1 Z0.6

In Action: The Printer stops at pause, after resume the printer moves over the printed Area, without moving to the right Z-Height, so he starts printing in the air.

**Expected results**
Like in all my previouse files the printer should start printing on the right Z-Height after pause

**Project file**
[flat30_32_con.zip](https://github.com/Ultimaker/Cura/files/4188077/flat30_32_con.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4188087/cura.log)

**Additional information**
Getting the issue in 4.2.1, upgrade to 4.4.1 doesn't helped (so the error Report from above is from the new 4.4.1) but the issue appeared first in 4.2.1.

Kayu84, 2020-02-11:
i've checked 4.5 beta, same issue

mahtDFR, 2020-02-13:
I've replicated this, and it is indeed a bug in the script.

We discussed as a team and can't dedicate our resources to this now so I've deferred it. We also don't have machines running repetier available here so it's difficult to test/support.

Feel free to make a pull request. We did update the generic pause at height script recently so this can be used as a basis, but note that repetier firmware works differently.

EEEvilBill, 2020-02-13:
When will this be addressed?  I have dozens of prints waiting that I suddenly can't do.

EEEvilBill, 2020-02-13:
I tried rolling back to 4.0 and the problem still exists.

EEEvilBill, 2020-02-13:
Roll back the generic script so that it at least worked the way it did two days ago.  An update is useless if it suddenly provides less functionality than before the update.

Kayu84, 2020-02-13:
> I've replicated this, and it is indeed a bug in the script.
> 
> We discussed as a team and can't dedicate our resources to this now so I've deferred it. We also don't have machines running repetier available here so it's difficult to test/support.

Well,  thanks for the response, I was really struggling about myself, doing anything wrong. 

I would like to help 😅 but I think I can't. Btw. I think you don't need to get a repetier for testing, because the issue is located in the gcode, not on the repetierserver himself.
But when ever you think you fixed that bug, I'm ready for test.

And I moved over to the 'pause at height' script and yes... while the 'M0' command paused the print, you can't ever resume.
Search and replace 'M0' by '@pause' works for repetier. Finished one of my six prints today.

EEEvilBill, 2020-02-14:
Thanks for your response.

I have no problem with the pause.  The resume works fine, but the bed by then has sunk and the nozzle is now so far above the print that it's just dumping filament into thin air.  The Z-axis needs to lock in place during the pause.

fieldOfView, 2020-02-14:
I'm willing to help out here, but I don't fully understand the issue. I don't have a repetier printer so I can not test this. Specifically, the screenshot shows 3 gcode files where - if I am not mistaken - the left one is somehow "correct" and the other two are not? Were they sliced with different settings?

How can I reproduce this?
* load an object (any specific model?)
* add "Pause at Height for Repetier" script (with what settings?)
* inspect gcode (what should I be looking at, what should be added/is missing where, exactly?)

Caveat: I can probably help getting this fixed, but the fix will not likely be in Cura 4.5 because that is feature-frozen, and this issue is not a regression introduced with Cura 4.5.

mahtDFR, 2020-02-14:
> Roll back the generic script so that it at least worked the way it did two days ago. An update is useless if it suddenly provides less functionality than before the update.

Just to be clear here: updates were not made to the repetier variant of the script, so it is the same as previous releases. 

[image]
****

Updates were made to the generic 'Pause at height' script. That's the only one we will actively support since our machines don't use repetier firmware. Details of the commit are here https://github.com/Ultimaker/Cura/commit/432ed02aa6eac01626b446777d6ae597ae2f23c3

EEEvilBill, 2020-02-14:
I have an Ender 5.  I edit the Gcode using the pull down menu.  "pause at height."  Select the number of layers to pause at.  The printer pauses at that layer, but the bed doesn't lock in position, it continues to slowly lower due to gravity.  When the print resumes, the nozzle is no longer at the correct height and is pushing filament into thin air above the print.

EEEvilBill, 2020-02-14:
I used the basic "pause at height"  for my Ender 5, the same as I have been doing for months.

[generic]<https://user-images.githubusercontent.com/17821912/74536862-35a5a600-4f39-11ea-83cb-242ffc5cade3.jpg>

Updates were made to the generic 'Pause at height' script. That's the only one we will actively support since our machines don't use repetier firmware. Details of the commit are here 432ed02<https://github.com/Ultimaker/Cura/commit/432ed02aa6eac01626b446777d6ae597ae2f23c3>

EEEvilBill, 2020-02-14:
(I'm on the Ender 5)

EEEvilBill, 2020-02-14:
Okay,  Did a full uninstall/reinstall of Cura.  Three times, it said there were code errors during setup.  Finally seems to have resolved the setup on the fourth try.  Did a pause at height test print and it seems to be working properly again.

fieldOfView, 2020-02-15:
@EEEvilBill I don't understand; the original issue reported by @Kayu84 is about the repetier-version of Pause at Height. If you are using the "normal" version of Pause at Height, it seems you were not having the same issue. But it is good to know your issue seems to be fixed.

@Kayu84's original issue also seems to be fixed by using the normal Pause at Height version and an additional Search and Replace to replace any `M0` with `@pause`, right?

Some day someone (...) is going to have to unify all these Pause at Height versions into one maintained version that selects what flavor to use automatically.

Ghostkeeper, 2020-10-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/5085.

closed 2020-10-14 by Ghostkeeper