Ultimaker/Cura #7093: [4.5-BETA] Cura looks like it is idle yet my fans are working their *** off

Appesteijn, 2020-02-12:
**Application version**
Latest 4.5-Beta. Previous version 4.1 installed.

**Platform**
MacOS 10.15.3

**Printer**
Network UM3

**Reproduction steps**
1. Start Cura
2. Login with account
3. Click Manage account

**Screenshot(s)**
[image]

**Actual results**
Cura eats all my CPU!

**Expected results**
Leave my CPU alone

smartavionics, 2020-02-12:
Yes, with my master branch I found that Cura was using 100% CPU after I logged in. Not sure of the sequence of actions though so I can't say exactly what happened.

[image]

mahtDFR, 2020-02-13:
Couldn't reproduce this in Windows, but I'll put it up for discussion. (Hi Mark!)
@smartavionics can you confirm the distro you're running? Ubuntu?

smartavionics, 2020-02-13:
Yes, I'm using Linux.

mahtDFR, 2020-02-13:
Thanks for the report. Internal ticket: CURA-7214

Ghostkeeper, 2020-02-13:
I'm not able to reproduce this on Ubuntu 19.10 either.
[image]

In my task manager it shows that PyCharm is doing something in the background, but Cura rests around 0-10%.

Do you have a log file? Maybe that would tell us what it's doing.

Appesteijn, 2020-02-13:
2020-02-12 09:12:57,428 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FileLogger
2020-02-12 09:12:57,439 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LegacyProfileReader
2020-02-12 09:12:57,450 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin AMFReader
2020-02-12 09:12:57,462 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SimpleView
2020-02-12 09:12:57,474 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SolidView
2020-02-12 09:12:57,492 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdateChecker
2020-02-12 09:12:57,501 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SentryLogger
2020-02-12 09:12:57,516 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin SupportEraser
2020-02-12 09:12:57,525 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin PrepareStage
2020-02-12 09:12:57,528 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [369]: Loading all plugins took 1.4682300090789795 seconds
2020-02-12 09:12:57,537 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [148]: Looking for old configuration files to upgrade.
2020-02-12 09:12:57,568 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_left+%232_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,571 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_right+%232_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,574 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./UltiMacher_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,577 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./Ultimaker+S5_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,580 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_right+%232_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,583 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_left+%232_settings.inst.cfg] with type [definition_changes] and source version [4000007]
2020-02-12 09:12:57,605 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_left+%232.extruder.cfg] with type [extruder_train] and source version [4000007]
2020-02-12 09:12:57,607 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_right+%232.extruder.cfg] with type [extruder_train] and source version [4000007]
2020-02-12 09:12:57,609 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_right+%232.extruder.cfg] with type [extruder_train] and source version [4000007]
2020-02-12 09:12:57,612 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_left+%232.extruder.cfg] with type [extruder_train] and source version [4000007]
2020-02-12 09:12:57,624 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./UltiMacher.global.cfg] with type [machine_stack] and source version [4000007]
2020-02-12 09:12:57,626 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./Ultimaker+S5.global.cfg] with type [machine_stack] and source version [4000007]
2020-02-12 09:12:57,636 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./cura.cfg] with type [preferences] and source version [6000007]
2020-02-12 09:12:57,661 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_left+%232_user.inst.cfg] with type [user] and source version [4000007]
2020-02-12 09:12:57,664 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker_s5_extruder_right+%232_user.inst.cfg] with type [user] and source version [4000007]
2020-02-12 09:12:57,666 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_left+%232_user.inst.cfg] with type [user] and source version [4000007]
2020-02-12 09:12:57,669 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./UltiMacher_user.inst.cfg] with type [user] and source version [4000007]
2020-02-12 09:12:57,671 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./ultimaker3_extruder_right+%232_user.inst.cfg] with type [user] and source version [4000007]
[... cut ...]
2020-02-13 13:03:37,340 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [461]: Shutting down cura
2020-02-13 13:03:37,378 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [292]: Attempting to kill the engine process
2020-02-13 13:03:37,379 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [298]: Killing engine process
2020-02-13 13:03:37,379 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Engine process is killed. Received return code -15

Ghostkeeper, 2020-02-14:
Thanks! But it seems that the log file agrees that Cura is idle. It's not listing that it's busy doing something. :cry: 

We're keeping an eye out, but haven't seen this happen yet.

One more thing we can try is to temporarily disable the UM3NetworkPrinting plug-in in the Installed tab of the marketplace. We've had in the past that Zeroconf was taking a lot of processing power. We've recently made a fix to Zeroconf as well to reduce processing power. If disabling that plug-in helps for you, there's a good chance that the 4.5 release will fix it for you.

nallath, 2020-06-18:
This should be fixed in the 4.7 release. Turned out it was the authentication server wasn't being closed properly (although it was told to shut down...) 

See https://github.com/Ultimaker/Cura/pull/7902

closed 2020-06-18 by nallath