Ultimaker/Cura #7094: [4.5 beta] no start or ending gcode field at the first start

Liger0, 2020-02-12:
**Application version**
4.5 beta

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Install cura
2. Select cr10s printer

**Screenshot(s)**
[image]

**Actual results**
No space or way to make gcode visible

**Expected results**
Space and bars to make gcode visible and the areas bigger

Vandrasc, 2020-02-13:
Hi @Liger0 , did you tried to maximize the window by dragging it? 
For me, the windows do show for both start and end gcode but the space is very small.

mahtDFR, 2020-02-13:
Thanks for the report. I couldn't reproduce this in Windows 10 x64
[image]

Any details about your screen resolution?

I also noticed when resizing the window and scaling it down again it produced some weird behaviour so I'll look into that.
[image]

mahtDFR, 2020-02-13:
[image]
@Liger0 What's your display scaling setting? Just tried it on 150% and it creates similar issues. I'll put it up for discussion on how to handle scaling above 100%

Liger0, 2020-02-13:
I use 125%. I tried to increase the screen or use full screen but no difference.

mahtDFR, 2020-02-17:
Thanks for the report. We discussed briefly and decided to defer this for now. We recognise it's a bug, but don't have the resources to cover it at this time.

Aardvarx55, 2020-03-22:
I have just installed a BLTouch to my Ender-3 Pro, but when I go to add the required G-code, the displayed areas for this are too small (Mac)
[image]

Liger0, 2020-06-19:
Reopen?

Ghostkeeper, 2020-06-30:
Indeed this is still a problem. For our planning we're waiting with a solution until we've removed the system theming from this dialogue.

closed 2023-06-13 by github-actions[bot]