Ultimaker/Cura #7096: [4.5 beta] Material goes unselected when changing brand

Liger0, 2020-02-13:
**Application version**
4.5 beta

**Platform**
Win10x64

**Printer**
Cr10S

**Reproduction steps**
1. Duplicate a generic 1.75 material
2. Click on the new material
3. Change the brand name and either press Enter on the keyboard or click on another field of the filament

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
The selection of the material disappears and you have to go find it again in the dropdown where it ended up to continue editing it

**Expected results**
Keep editing it

Ghostkeeper, 2020-02-17:
I'm unable to reproduce this issue. Here is a recording of the reproduction steps: [material_brand_rename.zip](https://github.com/Ultimaker/Cura/files/4214676/material_brand_rename.zip)

Maybe it's only with a specific material?

Liger0, 2020-02-17:
Updated with video and log

Ghostkeeper, 2020-02-25:
Hmm, I tried with the same printer and the same material to duplicate, and also starting from having ABS loaded in the printer, and for me the custom material stays selected :confused:

Liger0, 2020-02-25:
Weird. It only happens with 4.5 beta.

Ghostkeeper, 2020-02-28:
In 4.5 we've changed some signal emitting around to improve performance. This caused some bugs that we (hopefully) fixed in 4.5 stable. So maybe that's how it got fixed for stable, then, and that's why I couldn't reproduce it with my development version. Thanks for the bug report, anyway!

closed 2020-02-28 by Ghostkeeper

fieldOfView, 2020-02-28:
[image]

Liger0, 2020-02-28:
I have tried 4.5 stable and the behaviour is terrible in the material settings, it constantly freezes everything I do!