Ultimaker/Cura #7097: 4.4.1: Support Interface Partial Coverage

CCS86, 2020-02-13:
Running 4.4.1 on Windows 10 and noticed this small issue:

[image]

I have verified that the model's bottom face is parallel to this face, and even at a multiple of layer height. Subsequent face above this doesn't exhibit the same issue:

[image]

[gcode.txt](https://github.com/Ultimaker/Cura/files/4200749/gcode.txt)

[model.zip](https://github.com/Ultimaker/Cura/files/4200757/model.zip)

Ghostkeeper, 2020-02-17:
Hrm, I'm not getting that issue.
[image]
Could you try changing the Support Stair Step Height to 0? That has given us this sort of issues in the past.

Otherwise, can you send us a project file completely? Perhaps some of my settings are still different from yours.

CCS86, 2020-02-17:
Looks like you are looking at the second support interface floor, which I had no problems with either.

I can confirm that setting the support stair step height to 0, resolves the issue. Thanks for the workaround!

rburema, 2020-02-19:
If that workaround is the answer, then it's very likely a duplicate of a long list of issues.

Closing this because it's a duplicate, please continue any discussion here: #6379

closed 2020-02-19 by rburema

rburema, 2020-06-25:
We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.

CCS86, 2020-06-25:
> We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.

Sweet, thanks!