Ultimaker/Cura #7100: 4.5 Beta - Issue with "bridge sparse infill max density"

johnkray, 2020-02-14:
**4.5 Beta**
(The version of the application this issue occurs with.)

**Platform**
Windows 10

**Printer**
Hydra Research Nautilus

**Reproduction steps**
Enable Bridge settings.

**Actual results**
It appears that when bridge settings are enabled the new "bridge sparse infill max density" is ignored and bridging is calculated regardless of the infill percentage and threshold defined. Further, bringing is calculated for all skin layers except for the final top layer.

**Expected results**
I would think you would only want bridging calculations for the first layer on top of sparse infill and then normal skin layers after that. Or, if multiple bridge layers are defined follow those rules.

johnkray, 2020-02-14:
**Update**
I just switched between active printers a couple times and now everything seems to be working as expected. Not sure what was causing the issues. I will post again if it happens again.

johnkray, 2020-02-15:
**Update 2**
The issue is back and is also present when tested with the Ultimaker 3

closed 2020-02-15 by johnkray

Ghostkeeper, 2020-02-18:
Did you intend to close this? Was the issue fixed?

I've tested this feature and it seems to work for me. I wrote an article about it: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_sparse_infill_max_density.md

johnkray, 2020-02-18:
Yes, I did close this issues. I discovered that the issue was caused due setting "Bridge Skin Support Threshold" to 100%. I had done that trying to force bridge calcs for testing, didn't realize that adverse affect it would have.

My issue it turns out is more to do with how Cura detects/calculates bridging paths. I may start a new topic to discuss this.