Ultimaker/Cura #7103: 4.5 beta - "Nothing to show" popup

tlhintoq, 2020-02-15:
Please either get rid of this popup or give us a checkbox in the preferences to turn it off.

Its annoying as all get out to be in the preview and have it pop up every time you make a change.

There is already so little screen left to view the bed with all the pallets, and overlays adding this really makes it worse.

Or at the very least have it appear and AUTOMATICALLY dismiss itself after 3 seconds.

fieldOfView, 2020-02-15:
Just so we know what you are talking about, is it the message that appears in the Preview tab after slicing if there are no models that are totally inside the buildvolume? It actually says `Nothing to *slice* because none of the models fit the build volume or are assigned to a disabled extruder. Please scale or rotate models to fit, or enable an extruder.`

tlhintoq, 2020-02-16:
@fieldOfView 
That is correct
[image]

Liger0, 2020-02-16:
That is extremely annoying as it appears everything you do indeed.

Ghostkeeper, 2020-02-18:
We've decided to put a checkbox there to not show it again.

tlhintoq, 2020-02-19:
Outstanding.  Thank you !

BagelOrb, 2020-02-21:
If you click the pop-up away it will stay away even when making changes, *except* when you have Slice Automatically enabled.

However, when Slice Automatically is enabled the layers will automatically get loaded.

I therefore propose to never show the message when Slice Automatically is enabled.

BagelOrb, 2020-02-21:
Just implemented both my and @Ghostkeeper s idea.

Ghostkeeper, 2020-02-25:
Fixed for 4.5 stable.

closed 2020-02-25 by Ghostkeeper

alfrankgit, 2022-02-05:
Still there in 4.13.1 with automatic slicing disabled.
If you close it, "Processing layers" gets stuck at 72% (3DBenchy loaded).
No preview appears.

Ghostkeeper, 2022-02-19:
A log file could help to see what the issue is there then. I'm not able to reproduce that issue. Processing the layers completes as normal. Do you have any reproduction steps, perhaps a specific model that it occurs for, if it doesn't happen always?