Ultimaker/Cura #7104: Make pallets and overlays draggable off the main screen

tlhintoq, 2020-02-15:
There is so little actually working space.  Maybe 25% of the screen is used for seeing the bed.

Making all the pallets (Settings, tool selection, line type selector, object list) draggable so we can put them on a second monitor would be amazing.  I would give anything to dedicate a monitor just to viewing the print bed so I can zoom in and still see enough of the model to have some context of what I'm looking at.

Ghostkeeper, 2020-02-18:
It ends up at around 40% of the screen being used for the 3D view if everything is open and you have the minimum supported screen size. Especially the settings panel gets in the way then. But I'd think that if you have multiple screens, most likely your set-up is expensive enough that you have at least an 1080p monitor by now! Still, it's a valid point. Some observations about the current state:

* You can currently drag the settings list and colour scheme panels around. The settings list can be resized.
* The rest of the panels at the top (printer selection, configuration selection, view selection) automatically close when they lose focus so I don't think they are a problem.
* The tool panels can't be dragged but I think user experience wise that would be a bit weird since they are stuck to the left side as a slide-out panel there.
* The object list can be made to be dragged. It would need a background colour though for it to be sensible, which could make it look more cluttered. I don't feel like this is really obstructing anything though, but then again I don't usually expand it.
* The action panel (slice/save button and such) can be made to be dragged pretty easily I think.

What you're really asking though is to be able to drag those outside of the window. I don't know how hard that would be technically in Qt. Maybe some operating systems would require a window border around them.

tlhintoq, 2020-02-18:
Being dismissive to a user because having multiple monitors automatically means "they have a very expensive system" is not good customer service. Especially when monitors are only $100 and your average laptop can run 3 of them.

I could put it another way...
I use the same software development tools you probably do.  We both know just how much multiple monitors increase productivity.  Having to constantly drag, move, shrink, move, close, open, drag, shrink, zoom, move.... Is not way to be productive.

> if everything is open
Well of course everything is open.  I'm *working* here.  How is someone going to work in the application if everything is closed?  And as I said... Open, close, make changes, close, open, preview, close, preview, open make changes {...}  all day is not a good workflow.

> It ends up at around 40% of the screen being used for the 3D view
Exactly.  That's terrible for someone working in your application all day long.  Its probably fine for the guy doing a trinket off Thingiverse once a day.  But for those of us *working* in 3D shops full time it horrible.

I'm sure your developers have all their Visual Studio pallets spread out for the exact same reason: Just like I do.  You can't deny how much more productive it is.  

[image]

Same with all the Adobe Suite of products: You move the pallets to other screens so your artboard/workspace is full screen.  Its not an uncommon working environment.

mahtDFR, 2020-02-19:
A good feature request indeed! Unfortunately because of the amount of time it would take to implement this, there's nowhere to fit it in our roadmap right now, so I have to defer it. We might pick it up in future but there's no time to devote to it right now.

DonExo, 2020-02-25:
There is however a Plugin that does similar thing to what you are asking.
It's called Sidebar GUI. You should look it up in the Marketplace.

[image]

Here how it looks like with it

tlhintoq, 2020-02-25:
@DonExo 
Thanks.  Already running it for the last several versions.  It helps give back a few percent.  But you still have all the tools ON TOP OF the workspace: that's what I'm trying to get fixed.  Trying to get the tools off the work surface and onto another monitor so the workspace is just the actual print bed.

As soon as you have an object selected the left tools pop open their pallet another 100px into the workspace.  If you're in preview you loose another 35px for the layer slider.  Open the object browser and there goes another 150 pixel by the time you have the selector which is is _offset from the left margin by 30px_ wasting more space.

If you have Cura open and tools *in use* as one does all day, you're left with maybe 25% of your screen to actually see the printer bed, create your layout of multiple objects, figure out spacing between objects, watch the preview play large enough to see what matters while have enough surrounding area to have context so you can decide if there will be head collisions with other objects in one-at-a-time mode.

The point I'm making is this is fine for the casual whacker making flexi-lizard trinkets for their grand children.  For someone running a 3D printing *shop* and are in here continuously doing commercial-grade work this is VERY restrictive and claustrophobic.

Ghostkeeper, 2020-02-25:
> Being dismissive to a user because having multiple monitors automatically means "they have a very expensive system" is not good customer service.

I'm not customer service, sorry, but I was only making the assumption that the type of people who have 4 monitors are also the type of people that buy monitors with more than 720p resolution. Customer service would be your Ultimaker reseller. They also have experts that could help you find a solution.

We understand the need for a feature like this and we'd gladly get the resources of a company like Microsoft on it, but we don't have them so we're not going to give this any significant priority. The cost of implementing this is fairly high.

tlhintoq, 2020-02-25:
No offense but... Huh?
I've been a developer for 25 years.  It doesn't take any significant amount of time or money to make tool pallets that can move.  In fact you have to go out of your way to *stop* it from happening.  If your developers are telling you otherwise: Stop outsourcing to India.

Tool pallets are by their very nature their own objects and UI elements.  Again, unless someone has gone way out of their way to do otherwise and burned a huge number of hours to do so.

> I was only making the assumption that the type of people who have 4 monitors are also the type of people that buy monitors with more than 720p resolution.

And this is why you don't make those kinds of assumptions.  Companies order monitors to save money just like everyone else.  1920x1200 or 1920x1080 is the standard.  *People* can do pretty much what they like when its out of pocket.  *Business* hardware approvals are run by the accountants and if 40 other people in the company are fine on 1920x1080 for $200 then that's what everyone gets. 

Again - there is clearly a big disconnect between what I'm trying to describe as a business situation - and your perspective that all your users are individuals at home.  I find it sad that your company outlook is that your product is only good even for private parties and that you have no aspirations to be better - to be good enough for enterprise/company installations.

ChrisTerBeke, 2020-02-25:
> No offense but... Huh?
I've been a developer for 25 years. It doesn't take any significant amount of time or money to make tool pallets that can move.

Great! As Cura is open source, I'm sure you can make a nice PR to implement this then.

Ghostkeeper, 2020-02-25:
I think this is degrading more towards namecalling and establishing dominance now, and accusations of having Indians help us. Let's keep this on topic, because it's going on a tangent on a point that I think we are agreeing on: This feature would be useful.

The development team deems this difficult enough to outweigh the benefits given our other priorities, but if you can give us some pointers on where we went out of our way to lock up the tool panels within the window, we can reconsider. We just need some help then, since QML is not the easiest framework to implement this in. Have you ever worked with PyQt, @tlhintoq?

tlhintoq, 2020-02-25:
Sorry if you felt there was name calling.  I don't believe anyone called anyone a bad name.

I'm trying to tell a manager that if the team member is saying "its a bunch of work" then there is a problem.  ~Maybe that problem is with the original architecture because it **shouldn't be**, using any kind of modern environment from the last 10 years.~ Maybe its a problem with the team member being a jr. dev and just not knowing how.  We were all jr's once and maybe they just need some guidance.

Disregard the previous paragraph.  It doesn't apply.  You're not *in* a modern framework or environment at all.  **My mistake**.

QML?  I haven't heard that term in years.
So y'all wrote/write this with JavaScript as the backing... or C++ at best?

Never mind.  I withdraw all my comments about it shouldn't be a problem.  In fact I compliment you guys for having it work _at all_ if you're forced to work in that environment.  I make six figures a year with 20 years in full-stack development, currently doing cross-platform development for the last 6 years - and honestly you **couldn't pay me enough** to cross over into that world: I mean that with the _utmost admiration_ for the people you have doing this work.

You are absolutely right - trying to do even the simplest task there is a major undertaking.  I understand now why your people don't want to make big changes because its totally expected that even small changes will break everything all the pieces and cause a massive amount of regression testing and reworking.

fieldOfView, 2020-02-25:
I learnt something new today I didn't think I need: how to block a user on github:
https://help.github.com/en/github/building-a-strong-community/blocking-a-user-from-your-personal-account#blocking-a-user-from-their-profile-page

tlhintoq, 2020-02-25:
@fieldofview
Not sure what that was about.  My last comment was trying to praise their team for their accomplishments in a very trying working environment.

nallath, 2020-02-26:
> I've been a developer for 25 years.

I'm of the strong opinion that the statement "I've been a developer for x years" is a bit the same as the "I'm not racist but ...".  Somehow it looks like a good statement to make, but it pretty much always derails the rest of whatever you are saying after ;)

> @fieldOfView
> Not sure what that was about. My last comment was trying to praise their team for their accomplishments in a very trying working environment.

No, it wasn't. Don't try to pretend that it is. You're just lashing out because someone refused to implement a feature that you want. Instead of listening to what people had to say, you decided to start arguing about it, showing a complete lack of understanding of the situation (and being condescending, racist and rude about it). This is a really counter-productive thing to do, and you should no better since you claim to be a very well paid developer with eons of experience. QML (and Qt for that matter) is one of the _most used_ frameworks at the moment. Just because you didn't hear from it, does not mean it's not modern (it's your understanding that's a bit outdated perhaps?)

tlhintoq, 2020-02-26:
> I'm of the strong opinion that 

Well... As you said, that's your opinion.  On the other hand many people when receiving a compliment like to know that it comes from someone that's been in their same industry for a substantial amount of time.  Its nice to know a peer of your same experience thinks highly of you.

Please don't confuse the comment about outsourcing to India with racism.  There's nothing wrong with being Indian.  My boss is along with several coworkers. I have no issues there. But here we're talking about the country not the individuals. Outsourcing to India however brings with it a LOT of very well documents problems ranging from time zone issues to lack of supervision, out-dated skill-sets, to the nature of the outsourcing culture there where 10 interns are put on a task that one sr would do and google/copy/paste is the *expected* way to handle it. You know you see those posts on every developer board from MSDN to StackOverflow.   I couldn't care less about someone's race. There is however a massive _cultural_ problem for developers in the country because of how they are treated as disposable.  Its not right.  People shouldn't be treated that way.  But because it does happen their developers won't say "I don't know that" out of fear of getting fired.  Its a shame, and it is changing slowly.  But it is undeniably how things happen right now. Once these bright and talented people are taken out of that situation, brought to countries where they are up-trained and free to express where they need improvement they thrive. I've worked with a few and sadly they remind me of people from abusive relationships, scared to say anything. But once they know they are safe from retaliation they become exceptional team members.

As for debating QML/QT... That's a whole 'nother matter and one that would seriously hi-jack this thread.  I just wanted to clarify my position on the one point, as well as re-iterate that I was being sincere in my compliments to the team for what they've accomplished given my feelings on the coding framework.  You may not share my opinion of that framework, but at least accept that from my perspective of it - I feel they've done a lot.

tlhintoq, 2023-06-30:
Personally... Yeah I think it would make life a lot easier on the demographic of users that are on single screen systems... and screens of common size.  Which is probably a much larger percentage than those of us with Ultrawide monitors.
As you can see, here on a 2560x1080 ultrawide even a 430mm bed printer can still be seen in its entirety at a reasonable magnification to work and get a feel for the job you're doing.
[image]

Now look at the same print bed after dragging it to a 1920x1080 screen.
[image]

Even folks that have two monitors aren't getting any benefit because Cura only uses one.  **BUT** if you can drag the pallets off the main view and stick them to the side even if its just another 1920x1080, then the main viewing area is freed up for your print bed viewing.

Here we can see a program called Armorsmith which allows the dragging of it's tool pallets.  Its a huge difference between a single crowded screen, and using one monitor for viewing and the other for pallets.
[image]

Just one guy's opinion.