Ultimaker/Cura #7105: Support infill ignores inner boundaries when using tree support

Jmlannan, 2020-02-15:
When you have support going all the way around a print and have support density not zero cura will try to fill in the support infill all the way through the print. 

**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Tevo tarantula

**Reproduction steps**
1. Starting with a part with an overhang that goes around the part or will have support going around the part
2. Turn on tree support with nonzero support density
3. Check preview to see that support infill is now going through the part.

**Screenshot(s)**
[image]

**Actual results**
Support infill joined to print

**Expected results**
Support infill should only be within the supports not in the print.

**Project file**
[TT_Vive_Katana_Handle_Adapter.3mf.txt](https://github.com/Ultimaker/Cura/files/4208991/TT_Vive_Katana_Handle_Adapter.3mf.txt)

Ghostkeeper, 2020-02-18:
You spent a lot of time optimising this print, I can tell.

That bug indeed reproduces. I feel like I've seen this before but I thought we fixed that.

mahtDFR, 2020-02-19:
This is on our backlog as CURA-7236

Ghostkeeper, 2020-03-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/5223.

closed 2020-03-10 by Ghostkeeper

nallath, 2020-07-01:
Fixed for 4.7