Ultimaker/Cura #7107: interface problem (Сura 4.0 - 4.5)

Goodfeat, 2020-02-16:
Hello. there is one small problem that is present in all versions of the program

in order to fit as many options as possible in the window, I use a custom theme

but the problem is that not all configuration cells look the same,
can this bug be fixed?
[image]
[image]

Ghostkeeper, 2020-02-25:
Seems to be something with the `NumericTextFieldWithUnit.qml` file we wrote. It's using a `TextField` element which has the default of `verticalAlignment: TextInput.AlignVCenter` which is supposed to align the text vertically. However it doesn't do this properly; clearly the text is aligned too high. Problem is though that this QML element doesn't allow customising the label in any way other than with its own fields (like `verticalAlignment`).

We just have an extra label for the unit in there that is put inside the TextField by ourselves and aligned with its centre. The label for the unit is working properly.

I couldn't find a quick solution for this. We have seen similar things before where users reported that the text was hard to read in some pixel scales.

smartavionics, 2020-02-25:
The same problem occurs even if you don't use a custom theme. This is what I see on Linux.

[image]

Goodfeat, 2020-02-25:
There is a temporary solution that I have to use
its essence is that you need to add a value to an existing variable in the NumericTextFieldWithUnit.qml file

I just open the file NumericTextFieldWithUnit.qml
and add to the string another value to the string
[image]

this way I have to correct the code for my style every time a new version of the program comes out

closed 2023-06-13 by github-actions[bot]