Ultimaker/Cura #7111: Cura 4.4.1 - Profile and Printer settings/values sometimes (not always) revert back

rplant45a, 2020-02-16:
**Application version** - 4.4.1
(The version of the application this issue occurs with.)

**Platform** - Win 10 
see image attached

[image]
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer** - Creality CR-10
(Which printer was selected in Cura?)

**Reproduction steps**
nothing specific, I changed 
1. from skirt to brim and when I loaded Cura again it had reverted back to skirt.
2. start gcode by changing some for EZABL code to raise hot end higher before it went to home and printed the two lines to clear the nozzle. Also changed the z offest.  Both of these gcode changes revered back to previous instances.

**Screenshot(s)** - I don't think it's needed but if you want a shot of my gcode I'll be happy to supply but it's just the EZABL with two insignificant changes
(Image showing the problem, perhaps before/after images.) 

**Actual results** - like I said cura reverted back to previous instances of the about 2 changes
(What happens after the above steps have been followed.)

**Expected results** - should not have reverted
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-02-25:
I'm suspecting that maybe Cura didn't close properly and wasn't able to save its settings? You could try this by waiting for 10 seconds after you made the setting change, as it should autosave every 10 seconds. You can also verify that this happened by looking in the log for the entry "Autosaving preferences, instances and profiles".

I haven't been able to reproduce this. I don't think it's specific to a certain printer or profile, as our code tends to be generic for that sort of thing.

Vandrasc, 2020-03-16:
I am also not able to reproduce this issue. @rplant45a are you also experiencing this in the latest Cura 4.5?

rplant45a, 2020-03-16:
No has did not happen again after I created the post.  But like I said it happened a couple time too many.  But, I did upgrade to 4.5.1 and have not had any issues at all.  Could be that I indeed crashed the instance and it reverted back to pre profile… being a sw person myself, I could see how this could happen and be a fail safe… and if the case, it would be good to have msg about trouble reading current profile (file read error) and reverting to possible know good profile… but thanks for checking!!!!!!!!!!!  You guys are doing such a great job and I know it’s probably on your own time!!!!!!!!  So thanks a bunch!!!!!!!!!

 

 

Thanks…  Rick

 

Richard Plant

919-349-7890

104 Parktree Court

Cary, NC 27519

ישו הוא מלך

Ghostkeeper, 2020-03-18:
Let's assume the issue is fixed in 4.5 then. I think it wasn't a file read error but a file write error or simply Cura being interrupted before being able to write the file, so we wouldn't be able to show any error message then.

closed 2020-03-18 by Ghostkeeper