Ultimaker/Cura #7113: Cura 4.4.5 beta

pawien, 2020-02-16:


mahtDFR, 2020-02-17:
Thanks for the report. Can you attach a .3mf file that reproduces the problem?

* In Cura, click File > Save...
* Zip the 3mf
* Attach it here

Then we can test it

mahtDFR, 2020-02-17:
Thanks for the report. Can you attach a .3mf file that reproduces the problem? (In Cura go to File > Save...) then .zip it and attach it here.
Thanks for the report. Can you attach a .3mf file that reproduces the problem?

* In Cura, click File > Save...
* Zip the 3mf
* Attach it here

Please include a [log file](https://github.com/Ultimaker/Cura#loggingissues) if you haven't already.

pawien, 2020-02-17:
[GA_test1.zip](https://github.com/Ultimaker/Cura/files/4216143/GA_test1.zip)

pawien, 2020-02-17:
[GA_test1.zip](https://github.com/Ultimaker/Cura/files/4216149/GA_test1.zip)

closed 2020-02-17 by pawien

Ghostkeeper, 2020-02-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/6909