Ultimaker/Cura #7115: [4.5 beta] Rendering issue and crash

Liger0, 2020-02-16:
**Application version**
4.5 beta

**Platform**
Win10 x64

**Printer**
CR 10S

**Reproduction steps**
1. My profile selected
2. Enabled adaptive layers, disabled supports
3. Sliced

**Screenshot(s)**
D:\File personali\3d printing\problemi\Cura

**Actual results**
Rendering issues, crash

**Expected results**
No rendering issue, no later crash

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/4210883/cura.zip)

Liger0, 2020-02-18:
It has happened again. I have saved the project just in case, though I doubt it could be of any use. 
If it is, please tell me who to send it privately.

fieldOfView, 2020-02-18:
If you slice the same model/ settings again, does the issue reappear?

mahtDFR, 2020-02-18:
@Liger0 Email sent to you. Can you reply to it with an attached 3mf?

Liger0, 2020-02-18:
It doesn't seems to appear everytime I slice.
Project sent.

mahtDFR, 2020-02-19:
Hi @Liger0, thanks for the 3mf. I couldn't reproduce the issue as you described it in Win10 x64,  and it seems there's some extra parts of the model missing in the file you provided.

This is what I have:
[image]

We suspect the issue might be down to your graphics card. Is there another PC you can test it on with similar specs?

Maybe you can send another 3mf with the complete model as well? You can send it to the same address as before.

fieldOfView, 2020-02-19:
Didn’t @smartavionics see the same glitch after @nallath comitted some optimisations to layer view pre-beta?

Liger0, 2020-02-19:
The project I sent is the second one I posted the image of.

smartavionics, 2020-02-19:
Yes, I do sometimes see a glitch that has only started appearing recently but it looks different...

[image]

smartavionics, 2020-02-19:
Actually, I haven't seen it for a couple of weeks so perhaps they have fixed it.

Liger0, 2020-02-19:
May the defective layer where the rendering line goes to be the one that's referred  in the log?

`2020-02-16 21:29:25,444 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING] Layer 453 is empty (or it has empty extruder plans). Temperature control and cross layer travel moves might suffer!`

nallath, 2020-02-20:
> May the defective layer where the rendering line goes to be the one that's referred in the log?
>
> `2020-02-16 21:29:25,444 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [...]

I don't think so. In that case, a project would have issues regardless of the machine it ran on.

closed 2020-06-19 by no-response[bot]