Ultimaker/Cura #7116: Changing Object Names on Object list

valeria-gomez, 2020-02-16:
**Is your feature request related to a problem? Please describe.**
Hello Cura team. Thank you for the free software that makes our work possible! I participate in a research in our university to study the impact property of different mesh fill density and wall thickness for different applications.  

Using Cura, me and a small group of researchers load multiples of the same part and uses the feature "Per Model Settings". Every part on the bed have particular sets of settings. When working, there are at least 20 parts on the workspace. Every one part has a different value for density, wall count, and other settings that we experiment on. 

We found the Object List to be useful. We would like to rename the object names. If we can update the name of the objects, it will help improve our siatuations. 

Each part on the bed have different settings. To not forget which part has a particular setting, we track it using an excel sheet. 
[image]

Another situation in which we see this feature to be useful is when we have multiple part that are intersected with other meshes that have additional "per model settings" with different values. With many different parts, we intersect it with several other models with their own "per model settings"

**Describe the solution you'd like**
Because the loaded part filenames are all the same, but each is modified differently in Cura, changing the name on the object list will help identify the parts easier. If this is possible, we can modify the list to indicate what the object does. 

**Describe alternatives you've considered**
We are currently using external excel sheets to keep track of the parts. 

**Affected users and/or printers**
It does not matter. If they use Cura, they will benefit from this.

rburema, 2020-02-25:
This is on our internal backlog as CURA-6798 (not to be confused with the github nr.s). We'll (hopefully) get to it in due time (this might take some time, our backlog is large).

Ghostkeeper, 2020-02-25:
Yeah this was also found by our UX engineer. It's a nice quality-of-life thing.

GregValiant, 2024-11-25:
This is now possible with the Mesh Tools plugin although it requires right-clicking on the model and then selecting "Mesh Tools" and "Rename Model".
Is this request still relevant or can it be closed?
[image]

closed 2024-12-09 by github-actions[bot]