Ultimaker/Cura #7118: [3.4.0] Horizontal Expansion = -0.1 mm is overly reducing thin walls/details

humphrey-b52, 2020-02-17:
**Application version**
3.4.0

**Platform**
MacOS Sierra 10.12.6

**Printer**
Prusa MK3s

**Reproduction steps**
1. I use "Horizontal Expansion = -0.1 mm" as a standard to gain dimensional accuracy.
2. I am aware that small feature can disappear, but it seems like the function reduces more than necessary.
3. Use case were letters on top of a print (screenshot no.1). Letters have a width of roughly 0.65 to 0.7 mm. Reducing them by 0.1 mm should not have the effect shown in screenshot no. 2. Making the letters unprintable.
4. Thought I am clever and reduced my line width (0.45 mm to 0.25 mm), but that did not bring the numbers back. (screenshot no. 3) - Why?
5. Solution for that specific print: Set "Horizontal Expansion = 0"

**Screenshot(s)**
Please see attached files (3)
[image]
[image]
[image]

**Actual results**
The small features were overly thinned, making them unprintable.

**Expected results**
The numbers being minimally thinner, -0.1 mm is not that much. 

**Project file**
GitHub doesn't support 3mf files.
Find it here: https://we.tl/t-vqVt8VZ5Y2

Ghostkeeper, 2020-02-25:
Your WeTransfer link has been degraded in the meanwhile, I'm sorry.

Github does support .zip files so you could rename the file (or just zip it up).

But perhaps first it'd be wise to test this in a more recent version to see if we have fixed it in the meanwhile anyway. I'd indeed expect that reducing the line width would help. But be aware that horizontal expansion reduces the thickness from both sides, so a thin wall would become 0.2mm thinner by setting the horizontal expansion to -0.1mm. 0.2mm is half a line width, typically.

Ghostkeeper, 2020-02-25:
I tried this with a model of an 1mm-thick wall. Setting the line width to 0.4mm and the horizontal expansion to -0.1mm made the lines match up exactly, as the wall is now 0.8mm wide:
[image]
Seems to be working correctly as far as I can see, so it would help to see that 3MF file.

closed 2020-06-19 by no-response[bot]