Ultimaker/Cura #7119: [4.5.0 Beta] Blocking "Ironing" feature by an Eraser box ruins main meshes

RealMook3d, 2020-02-17:
**Application version**
4.5.0 Beta

**Platform**
Win7 64bit

**Printer**
Custom (Tevo Flash)

**Reproduction steps**
1. Placed a mesh to slice
2. Placed a support blocker
3. Switched the blocker's type to change slice settings in this area
4. Choose "Iron" in the modification settings (for the same "Eraser" cube)
5. Unchecked "Iron" in the mod. settings (the sliced model shows that there is no ironing in the Eraser's volume - correct!)
5. Unchecked "Infill Only" - So I suppose that only "Iron" is modified.

**Screenshot(s)**

Sorry, no screenshots. It's easy to reproduce. Looks like it forms another mesh inside your model.
And overlaps in a weird manner. Maybe a boolean operation involved and calculates a model's mesh's polygonal walls wrongly.

**Actual results**
Got a damaged mesh which leaded to a broken sliced results (holes!)

**Expected results**
The actions should remove Ironing from the main model in the "modified" (settings) volume (the "Eraser" which then used to change settings, not to erase supports). 

**Project file**
Sorry, not yet. Will add if it's critical.

mahtDFR, 2020-02-17:
Thanks for the report. Can you attach a .3mf file that reproduces the problem?

* In Cura, click File > Save...
* Zip the 3mf
* Attach it here

Please include a [log file](https://github.com/Ultimaker/Cura#loggingissues) if you haven't already.

Ghostkeeper, 2020-02-25:
The issue can be easily reproduced with the reproduction steps. But here is a project file if necessary: [ironing_cutting_mesh.zip](https://github.com/Ultimaker/Cura/files/4249977/ironing_cutting_mesh.zip)

This is, currently, expected behaviour. The "modify settings of overlap" per-model setting breaks up the model into multiple models, and then allows changing per-model settings of the extra model. Breaking up the model into multiple models also inherently causes the contours of the walls to break up, the skin lines to not match up any more, etc.

This is sort of inherent in how this was designed. I'll propose this to the team to see if there is anything that can be done to change this, maybe they have an idea. But I don't have high hopes here.

RealMook3d, 2020-03-10:
Hi, guys!
Sorry, I wasn't able to answer earlier! I hope that it will be put in the road map at least.
Some places (or layers) need Ironing and some not. In many cases this saves a lot of printing time.
The same thing with the "resolution". If I need to print fine details (some text for example) in one single place then it would be the real time saver to make just a single small volume printed with 0.3mm lines (0.4 nozzle). I've did that trick already but it would be cool if it was supported officially.

mahtDFR, 2020-03-13:
Related ticket on our internal system: CURA-7255