Ultimaker/Cura #7120: Skirt Issue: Skirt printing under models

pmcsr, 2020-02-17:
**Application version**
4.5 beta (but was doing the same thing with the previous version)

**Platform**
windows 10

**Printer**
Prusa MK3s

**Reproduction steps**
1. Reloaded models
2. rearranged models and re-sliced

**Screenshot(s)**
[image]

Ghostkeeper, 2020-02-17:
Do you have a project file that reproduces this issue? It doesn't happen with any model, I suppose.

bergie5737, 2020-02-19:
Hi. I have just got the same problem. I cleared build plate. Imported a model. Multiplied it and saw skirt under the parts.
[image]

I printed the exact same footprint several times today with small tweaks to the height, and they printed with no problem. 
Restarting solved it for me.

smartavionics, 2020-02-19:
This will occur if the bottom of the model is not completely flat or the model is not aligned with the build plate. In such cases, a quick fix is to drop the model down by just enough so that all of the base is touching the build plate.

bergie5737, 2020-02-19:
Mmmm. I never checked that it was not flat so cannot rule it out with
absolute certainty.  I cleared the build plate and imported again and even
pulled a larger version in.  It's unlikely that I made the same mistake
again by not laying it flat. As soon as I restarted Cura again, I was able
to lay my model flat again...😁

I will check if this is the case next time I use Cura. On skirts, I had an
issue (that I did not report as I suspect I set something wrong,) where a
skirt near the edge of the build plate refused the model to be sliced -
which I understand if it goes out of bound. The skirt was under an overhang
that was wider than the skirt. Just removing the skirt, and the model
sliced. I thought at the time that the skirt gets put down wider than the
overhang even though the model itself fitted the build plate. The actual
contact area on the built plate was about 20mm inside the overhang.

bergie5737, 2020-02-19:
In the hope it helps by adding adding more...
I have deliberately not laid the model perfectly flat, then the skirt became a problem as stated.  (And I know that) I use "snap rotation" which for me is 15 deg at the minimum, and at 15 deg it is pretty obviously not laying flat. 

The other issue I describe where the skirt causes failure to slice, I can reproduce. I am attaching a screenshot. As you can see the shadow shows a much wider build area than the skirt. I can put one skirt line down, as soon as I add more, it shows my model is too big for the build plate.  The brim behave as expected up to where it eventually run out of real space.
Feel free to move or request me to log a separate issue for this behaviour.
[image]

Ghostkeeper, 2020-02-25:
That area is also affected by:
* Horizontal Offset
* Horizontal Offset Initial Layer
* Mold Width

And in one-at-a-time mode:
* Everything related to the bed adhesion
* Travel Avoid Distance
* The print head shape

Dragon12764, 2020-02-29:
I had to turn skirt off as it was making the skirt a completely different shape with part of it under the model.
[image]

Ghostkeeper, 2020-03-04:
For you also then: Do you have a project file that reproduces the issue?

Dragon12764, 2020-03-05:
No it was the model not the software, sorry.

fleutot, 2020-05-28:
Did you use "Select face to align to the build plate"? This [issue](https://github.com/Ultimaker/Cura/issues/7839) might be related.

closed 2020-06-19 by no-response[bot]