Ultimaker/Cura #7124: [4.5.0 beta] Initial Layer Bed Temperature fooled by comments

jmelquist, 2020-02-17:
**Application version**
v4.5.0 beta

**Platform**
Windows

**Printer**
Ender 3 Pro

**Reproduction steps**
In Machine Settings -> "Start G-Code", I have an M140 / M190 defined which over-rides Cura's temperature settings (that's fine/expected behavior).  However, when I comment out these lines (with a ";" char) Cura still doesn't not insert it's own bed-temp settings - specifically the "Initial Layer Bed Temp".  

**Screenshot(s)**
n/a

**Actual results**
When I remove the commented M140 / M190 lines, then Cura correctly inserts the bed temperature settings, including the initial layer temp and the non-initial layer temps.

**Expected results**
I'm assuming there is a substring search of [the user-defined start/end gcode] for M140/M190 and if found, it doesn't insert this command (since the user is overriding it) but Cura needs to ignore comments when this search is done (1. clean comments from gcode, 2. substring search).  ?possibly this is a larger problem with other settings?

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-02-25:
In fact it's searching for the string `{material_bed_temperature}`, `{default_material_bed_temperature}` or `{material_bed_temperature_layer_0}`. Not for `M140` or `M190`. In my opinion it should search for those M-codes instead, really. That would be more reliable if people are not letting it be filled in automatically by the setting values but hard-code their temperatures.

I think though that parsing through comments is perhaps acceptable since it offers a workaround if for some reason you don't want it to emit a heating command at all?

jmelquist, 2020-02-25:
Noted that **{*}** is the search condition, not the **M**code.  I casually agree with you on the codes being probably a better strategy over-all.

As for include/exclude comments during search:  I think it's non-intuitive to the user that general commented code would have an impact on Cura's behavior.  However, if you do ignore general Gcode comments, then the team might consider implementing a "Cura Directive" syntax... something like:
     **;#CURA#KEYWORD=VALUE(S)#**
For Instance, **;#CURA#OBJECTS_WITH_BRIM=1,2,5\r\n**  would...
 - be ignored by the printer (thanks to the semi-colon prefix)
 - be obvious to the user that it has a specific meaning to Cura
 - indicate to Cura that objects 1, 2 and 5 should be printed with a brim (but objects 3 and 4 should not be printed with a brim... a feature that 99% of users don't need)
This functionality could be used in many ways...
 - advanced users looking to override default cura behavior
 - allow beta features to be easily/quickly tested by the community w/o having to wrap UI around it.
 - when 99% of users don't need a feature, but 1% do have a valid need

Regardless, I appreciate the response and the consideration of the UserDirective idea.  thanks for a great product!  cheers!
-jas

nallath, 2020-02-27:
I've added a bit of code that ignores comments when checking if the heating commands need to be inserted. It's too late for the 4.5 release, but it will be in the 4.6 release.

closed 2020-02-27 by nallath

fieldOfView, 2020-02-27:
This change will break configurations where people are counting on the previous behavior.

Ghostkeeper, 2020-02-28:
> This change will break configurations where people are counting on the previous behavior.

That's basically what I said before here as well, yeah:

> I think though that parsing through comments is perhaps acceptable since it offers a workaround if for some reason you don't want it to emit a heating command at all?

However there is an argument to be made that this was then depending on a bug for its implementation. And the use case is very small.

fieldOfView, 2020-02-28:
So, what's the new official workaround for this case:
> it offers a workaround if for some reason you don't want it to emit a heating command at all

[image]

jmelquist, 2020-02-28:
+1 for the "Cura Directive" feature to prevent loss of functionality... oh wait, that my suggestion so I probably can't "+1" it...  +1 all the same :)

Ghostkeeper, 2020-03-02:
You can workaround this you can add a `M117 {material_bed_temperature}` or `G88 {material_bed_temperature}`? It'll depend a bit on the printer firmware.

commit 1a44aa6d (2020-02-27): Ignore comments when checking for temperature prepends

Fixes #7124