Ultimaker/Cura #7127: Error of two extruders (4.5 beta): the second extruder does not turn off

Goodfeat, 2020-02-18:
Hello. version 4.5 beta detected an error
when turning off the second extruder, it still remains active

[image]

when the second extruder is turned off, printing to the second extruders should not occur

but the program still thinks the second extruder is active

fieldOfView, 2020-02-18:
I see you are using my sidebar gui plugin. Could you check if this also happens if the Sidebar GUI plugin is disabled?

Goodfeat, 2020-02-18:
> Не могли бы вы проверить, происходит ли это, если плагин Sidebar GUI отключен?

I already checked, your plugin works well.
this error appears even if you disable your plugin

This error has been present for a long time. I think I already wrote about this, but only half the bug was fixed.

Goodfeat, 2020-02-18:
I made a video to analyze the error.
https://youtu.be/cRs62DMkL2I

fieldOfView, 2020-02-18:
The video clarifies the issue. The problem is that when switching to a profile that uses eg the second extruder for adhesion *while the 2nd extruder is disabled*, the 2nd extruder still gets used. In other words: when switching to a different profile, Cura should check if the new profile uses extruders that are currently disabled and if so, change the settings to not conflict with the disabled extruder list.

mahtDFR, 2020-02-19:
Thanks for the report. It's a valid issue, but we've decided to defer this for now, as we don't have the resources to fix it.

Goodfeat, 2020-02-19:
it's a pity, I thought by the release of version 4.5 of the official version, this problem will be fixed forever.
Will have to put up with it again

closed 2023-06-13 by github-actions[bot]