Ultimaker/Cura #7128: [Cura 4.5.0 Beta] crash while re-slicing

kr15uk, 2020-02-18:
Cura 4.5.0 Beta

macOS Catalina 10.15.3 (2019 13" MBP, i7, 16GB RAM (3GB free), plenty free space)

Open STL, slice, go to preview, change any settings (tried infilled settings and wall settings), re-slice, again change something, re-slice, basically do that couple of times (around 3x usually)
Crash...

Ta.

K

mahtDFR, 2020-02-18:
Thanks for the report. Just tried in Mojave but couldn't reproduce it. Maybe it could be related to the model you're using, can you send a 3mf and a log as well to give us some clues?

kr15uk, 2020-02-18:
Tried 3x different STL's (1x thingiverse, 1x fusion and 1x from grabcad) - same.
After removing "sidebar gui" plugin it takes like 5x setting changes till it crashes, related?! dunno but worth mentioning I guess.

Please see 3mf attached and controlled test log.
[cura.log](https://github.com/Ultimaker/Cura/files/4221148/cura.log)
[473.3mf.zip](https://github.com/Ultimaker/Cura/files/4221152/473.3mf.zip)

jimmburk, 2020-02-18:
I am running MAC with High Sierra and slicing the Proto-Pasta spool with spokes and making slicer common modifications (infill, support, layers)  I typically change 1 thing at a time slice and see if I like the result.. (repleat ad nusium) and between 4 and 5 changed 4.5.0 will crash.

Vandrasc, 2020-02-21:
We have an internal ticket for this one, devs see CURA-7202 for our reference. 

Thank you all for your feedback!

Vandrasc, 2020-02-21:
The fix will be included in the 4.5 stable release.

closed 2020-02-23 by kr15uk

Ghostkeeper, 2020-02-25:
Gave us quite a bit of trouble, this one. Glad that you found it because it's pretty serious!