Ultimaker/Cura #7129: Cubicon Style Plus

mcrompton, 2020-02-18:
I am trying to use Cura to slice files for my Cubicon Style Plus.  While there is a profile for this printer in Cura, it does not save in the .hvs file format required for the printer.

I would love to see the existing profile edited to allow it to save to .hvs.

Unfortunately, Cubicon's own software is Windows only.  There is a web-based solution that can crash with complex files and is somewhat inconsistent.  I can get Simplify3D to generate files nicely, but in a school environment, I'm not going to force a paid option on my students.

Mac users looking for a free option for prepping files for their Cubicon Style Plus will benefit as will any user looking to manage their Cubicon within a fleet of other supported printers.

I am attaching a working benchy file in .hvs format.

Thank you,

Marc

fieldOfView, 2020-02-18:
> I am attaching a working benchy file in .hvs format.

I can't find it.

You may have to zip it to be able to attach it to a post here. NB: attachments from emails are scrubbed; you have to post the file here, or upload it somewhere and post a link.

fieldOfView, 2020-02-18:
Found the mentioned benchy hvs in the forum:
https://drive.google.com/open?id=1csGeCuLwBcu5hrinkmDGvgR3fD_9Poz_

fieldOfView, 2020-02-18:
I'm answering on the forum, here:
https://community.ultimaker.com/topic/31234-export-file-formats/

mahtDFR, 2020-02-19:
Deferring this as we have no way to support or test this. As stated in the forum thread, a plugin is probably the best way to handle this. Community contributions welcome.

marcoburato, 2020-12-12:
I've got a Cubicon Style (not the new Style Plus).
The firmware it came with it used to require the "M911 3DP-310F" mentioned in the forum, but it no longer need that with the latest firmware. I can just rename the .gcode file produced by Cura to .hvs and it works fine.
Maybe this applies to the new Style Plus too.

Would it be possible to add support to override the output file extension on the printer configuration?

nallath, 2020-12-14:
It's possible, but you would need to make a new plugin (eg; one that essentially uses the g-code plugin to create it's results, but provides a different file extension)

closed 2023-07-13 by github-actions[bot]