Ultimaker/Cura #7130: Ender 3 Pro Support Material Contact

SynthwaveFox, 2020-02-18:
**Application version**
4.4.1

**Platform**
macOS 10.14.6 (Mojave)

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Put a file on the bed
2. Enable supports (any type)
3. Start printing.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
When done with a layer, the printer will travel to start the next layer, but the nozzle doesn't seem to move up before doing that, so it hits every piece of support that it printed.
**Expected results**
The printer should print normally without making contact with the supports.

**Additional information**
The printer nozzle hits the support material when moving to the start of a new layer.

rburema, 2020-02-25:
Can you attach the saved project? (Save, not export -- you'll have to zip it to get github to accept it.)

bobster22388, 2020-02-29:
I have noticed this same issue. I will try find the project I had that did it last. I even tried changing the avoid height to something higher like 2mm, but still had the issue. Thankfully didn't affect the supports, but always worried the nozzle was going to knock them over.

bobster22388, 2020-02-29:
[CE3_frame.zip](https://github.com/Ultimaker/Cura/files/4270060/CE3_frame.zip)

Ghostkeeper, 2020-03-05:
If you want the nozzle to move up (over) the previously printed parts, you need to enable [Z Hops](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_hop_enabled.md). For your profile that is by default disabled, it looks like.

closed 2020-03-05 by Ghostkeeper