Ultimaker/Cura #7132: [4.4.1] Support dots appearing in air not connected to model

SlightlyLeprousHamster, 2020-02-19:
**Application version**
4.4.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 64bit

**Printer**
Creality CR-10

**Reproduction steps**
1. Model loaded (https://www.thingiverse.com/thing:4051669)
2. Model sliced with tree supports (support settings in screenshot)
3. Dots appear (screenshot)
4. During print, extrusion does occur at these locations.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

[image]

[image]

[image]

**Actual results**
Model prints mostly ok, but with extrusions at the phantom dots.

**Expected results**
These dots should not appear.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[CCR10_Frost_elemental.zip](https://github.com/Ultimaker/Cura/files/4222560/CCR10_Frost_elemental.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-02-25:
I don't have a ready solution right now, but as a workaround, you can try to set 'Minimum Support Area' to a small value (instead of `0`).

Ghostkeeper, 2020-02-25:
I've seen this bug myself as well. It can even occur if you've disabled support interface completely, for some reason.

rburema, 2020-03-17:
I think this is a duplicate of #7050 since creality has support x/y distance set to > 0 could be a workaround as well to set x/y distance to 0?

Closing this as duplicate, please see the open original ticket for further discussion.

closed 2020-03-17 by rburema

nallath, 2020-07-13:
Fixed for 4.7