Ultimaker/Cura #7134: [4.4.1] Slicer creating additional structures and filling hole

hobiseven, 2020-02-19:
Hi.
I try to print this extruder bracket for my Alfawise U30, and the slicer creates weird additional structures under the location of the extruder motor, and also fills the filament hole ( 2mm long horizontal hole) half way.

[Filament_detector_support_slicing problems.zip](https://github.com/Ultimaker/Cura/files/4223807/Filament_detector_support_slicing.problems.zip)

None of the settings I could change made any effect. 
When rotating the part to be vertical ( filament hole is then aligned with Z axis) , the slicing is then OK, but why would you print suc a part oriented like that?  

I attach the STL, as well as the pictures of the 2 problems. 1 - the filament long 2mm hole is being filled by the slicer at multiple locations. 2 - the weird 45 degrees additional structure created by the slicer where the surface should be totally flat.

Can any of you help? 
Thank you.

smartavionics, 2020-02-19:
Meshtools says the model has problems....

[image]

hobiseven, 2020-02-19:
Thank you for the fast answer! Ok, then I guess the model is not well prepared. 
I will do my own! 

BTW, your PB4 works really well. Greetings from France!

smartavionics, 2020-02-19:
> BTW, your PB4 works really well. Greetings from France!

Ah good, glad to hear it. Do you know Mike Goodrich? He is a good friend and expert gyro balancer who lives near Le Havre. (He knows nothing about 3d printing!)

smartavionics, 2020-02-19:
When I sliced your model it didn't show the same problems as your images show. So I think that maybe with some tweaks to the profile settings you can get a decent print. Please zip or rename the .3mf project file and attach to this issue. Thanks.

smartavionics, 2020-02-19:
[image]
[image]

smartavionics, 2020-02-19:
Setting the horizontal expansion to 0.05 gets rid of that weird slot...

[image]

hobiseven, 2020-02-19:
Here is the 3mf file zipped... My profile might be really broken, as I am a rookie in 3D prints.
[CE3_GuideFilament+Detecteur_assemble_v2.zip](https://github.com/Ultimaker/Cura/files/4224163/CE3_GuideFilament%2BDetecteur_assemble_v2.zip)

smartavionics, 2020-02-19:
Thanks for the project file. I tried fiddling with the settings but I could not stop it from filling the tube. I think you need to get a better model or somehow fix this one.

hobiseven, 2020-02-19:
It will take me only a little bit of time to redo a full model in Solidworks.... I will do that. It is better this way rather than banging the heads against the wall.
However, Cura developpers might look in this broken model to get a more bulletproof slicer behavior...

hobiseven, 2020-02-19:
@smartavionics  Mark, it is definitely linked to the profile parameters. With the default ender3 0.2mm profile, it is correct.... See attached. 
[CE3_GuideFilament+Detecteur_assemble_v2.zip](https://github.com/Ultimaker/Cura/files/4227918/CE3_GuideFilament%2BDetecteur_assemble_v2.zip)

closed 2020-02-19 by hobiseven

Ghostkeeper, 2020-02-26:
MeshLab even crashes on the MC edge collapse filter on that model o.O