Ultimaker/Cura #7135: where can i get help with developing GUI

starr3333, 2020-02-19:
I am trying to modify the GUI part with QtCreator.

The qml file seems doesn't work in QtCreator.

Is there any visual development environment available？

[image]

Thank you.

mahtDFR, 2020-02-19:
We use Pycharm internally. For a visual option we sometimes use qtcreator

starr3333, 2020-02-19:
@mahtDFR 

Thanks Reply,

Did you mean to import the entire Cura project into the QtCreator and run it? Or just import the qml part into the Qtcreator and run it!

I used to try to put single qml into QtCreator and run it, but QtCreator cannot found UM module. 

(I usually adjust the qml code by hand, but it is very inefficient. Is it possible to drag new component into the GUI like using QtDesigner?)

[image]

nallath, 2020-02-20:
I don't use a visual IDE to create the GUI. The exception to this is when I'm creating a new re-usable component. In that situation, I tend to first build it in a mock project in QtCreator before moving it into the main code.

You could try to also have a look at gamma-ray (from KDAB). It has some tools that can help with visual designing, but it's a complicated bit of software to use.

Ghostkeeper, 2020-02-26:
The problem with this is that the two imports: Cura and Uranium, are imports from our main Python code base. These imports provide the QML with very important information such as the theme, which defines most of the sizes and colours for the interface. So basically you'd need to run Cura in the background and somehow link that to your visual designer. Since this is hard to achieve, we've only gotten it to work by manually replacing/mocking Cura every time it needs it.

In practice, I just work with a text editor (non-visual) and run Cura to test it. There was once a shortcut to refresh the QML (Shift+F5) but that feature broke a while ago.

closed 2020-03-05 by starr3333