Ultimaker/Cura #7137: Extend Parameters functionality by reusing

dombmann, 2020-02-19:
Hello everyone,

I was wondering how to reuse some functionality. Lets say we have a parameter that changes the print orientation (similar to the Infill Line Directions parameter of the Infill section). Could this parameters functionality be linked to another parameter as to e.g. change the print orientation for e.g. a raft ? 

So basically : Create new Parameter that does something that is already given as another parameter
Is there any method you devs have thought of before ? Or is this unexplored territory ?

Most important question:
Can you imagine this to be solveable only on the frontend ?

Sincerley, 
Dom

mahtDFR, 2020-02-19:
This is possible in the frontend, but there's no user interface for it. Technically you'd be making your own material definition that maps one setting value onto another setting. What did you intend on using it for? I think the development would would be substantial.

dombmann, 2020-02-20:
Thanks for the reply, I was suspicious about the behaviour if the interal generation of GCode does not get changed when "re"using a params fuctionality. I was most interested in the feedback if it would be possible using only the frontend aswell as a rough estimation of the development necessary. The intention was to change the printing orientation for generated structures such as support (similar to how it is done in the infill). 

I guess this can be closed and labeled deferred

Ghostkeeper, 2020-02-27:
There is no setting for the direction of the raft lines though, so there is no way to make CuraEngine change the direction of the lines right now, from the front-end. For support we do have a setting: Support Infill Line Directions. And that setting is also in the front-end so you can just change it in Cura.

If you're talking about giving one setting the same value as another setting, or otherwise make it depend on the other setting, this can indeed be done but not from the front-end. Many of the built-in quality profiles use this to make settings depend on other settings, like here:

https://github.com/Ultimaker/Cura/blob/ba0eb114f6395cf235e5c5f8f03ef0506e70f5bd/resources/quality/ultimaker3/um3_aa0.4_Nylon_Normal_Quality.inst.cfg#L19

It can also contain more complex formulas, basically arbitrary Python expressions. Like this:

https://github.com/Ultimaker/Cura/blob/ba0eb114f6395cf235e5c5f8f03ef0506e70f5bd/resources/quality/ultimaker3/um3_aa0.4_PLA_Fast_Print.inst.cfg#L15

We had an idea to allow formula input in the interface but this is quite a large feature and only for very advanced users.

dombmann, 2020-02-27:
> We had an idea to allow formula input in the interface but this is quite a large feature and only for very advanced users.

Did you further pursue this idea? Sounds very interesting.

Ghostkeeper, 2020-03-02:
No currently it's not on the planning.