Ultimaker/Cura #7138: Cura master OSX keep crashing

Gipielle, 2020-02-19:
Process:               cura [455]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.5 (4.5.0-beta)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [455]
User ID:               501

Date/Time:             2020-02-19 22:10:33.558 +0100
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        036A1C6C-C5C0-997E-686C-56BF7E5887A0

Time Awake Since Boot: 49000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0:
--> 
    __TEXT                 00000001086d1000-000000010887c000 [ 1708K] r-x/rwx SM=COW  /Applications/Ultimaker Cura.app/Contents/MacOS/cura

Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x00007fff68e027fa __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff68d193a2 raise + 26
2   libsystem_platform.dylib      	0x00007fff68eb442d _sigtramp + 29
3   ???                           	000000000000000000 0 + 0
4   GLEngine                      	0x00007fff3bf0b787 glDrawElements_GL3Exec + 300
5   PyQt5._QOpenGLFunctions_4_1_Core.so	0x000000131af592f2 meth_QOpenGLFunctions_4_1_Core_glDrawElements(_object*, _object*) + 178
6   nl.ultimaker.cura             	0x0000000108726535 PyCFunction_Call + 165
7   nl.ultimaker.cura             	0x00000001087a4ccf PyEval_EvalFrameEx + 25599
8   nl.ultimaker.cura             	0x00000001087a9d12 0x1086d1000 + 888082
9   nl.ultimaker.cura             	0x00000001087a4733 PyEval_EvalFrameEx + 24163
10  nl.ultimaker.cura             	0x00000001087a9d12 0x1086d1000 + 888082
11  nl.ultimaker.cura             	0x00000001087a4733 PyEval_EvalFrameEx + 24163
12  nl.ultimaker.cura             	0x00000001087a9d12 0x1086d1000 + 888082
13  nl.ultimaker.cura             	0x00000001087a4733 PyEval_EvalFrameEx + 24163
14  nl.ultimaker.cura             	0x00000001087a9d12 0x1086d1000 + 888082
15  nl.ultimaker.cura             	0x00000001087a4733 PyEval_EvalFrameEx + 24163
16  nl.ultimaker.cura             	0x00000001087a94b2 0x1086d1000 + 885938
17  nl.ultimaker.cura             	0x000000010879e8ae PyEval_EvalCodeEx + 78
18  nl.ultimaker.cura             	0x000000010870689d 0x1086d1000 + 219293
19  nl.ultimaker.cura             	0x00000001086dd0b5 PyObject_Call + 101
20  nl.ultimaker.cura             	0x00000001086f2bed 0x1086d1000 + 138221
21  nl.ultimaker.cura             	0x00000001086dd0b5 PyObject_Call + 101
22  nl.ultimaker.cura             	0x00000001087a8425 PyEval_CallObjectWithKeywords + 165
23  PyQt5.QtCore.so               	0x0000000113437238 PyQtSlot::call(_object*, _object*) const + 40
24  PyQt5.QtCore.so               	0x0000000113437136 PyQtSlot::invoke(void**, _object*, void*, bool) const + 374
25  PyQt5.QtCore.so               	0x0000000113437b28 PyQtSlotProxy::unislot(void**) + 88
26  PyQt5.QtCore.so               	0x0000000113437a9a PyQtSlotProxy::qt_metacall(QMetaObject::Call, int, void**) + 58
27  libQt5Core.5.dylib            	0x00000001137fa437 QMetaObject::activate(QObject*, int, int, void**) + 1959
28  libQt5Quick.5.dylib           	0x0000000818485db4 QQuickWindowPrivate::renderSceneGraph(QSize const&) + 116
29  libQt5Quick.5.dylib           	0x0000000818417abd 0x81836d000 + 699069
30  libQt5Quick.5.dylib           	0x000000081848c4fc QQuickWindow::event(QEvent*) + 460
31  PyQt5.QtQuick.so              	0x00000001179f95aa sipQQuickWindow::event(QEvent*) + 106
32  libQt5Widgets.5.dylib         	0x00000001164540ba QApplicationPrivate::notify_helper(QObject*, QEvent*) + 314
33  libQt5Widgets.5.dylib         	0x00000001164566ed QApplication::notify(QObject*, QEvent*) + 5181
34  PyQt5.QtWidgets.so            	0x0000000115f2dd62 sipQApplication::notify(QObject*, QEvent*) + 114
35  libQt5Core.5.dylib            	0x00000001137c6ae4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
36  libQt5Gui.5.dylib             	0x00000001159ed136 QWindowPrivate::deliverUpdateRequest() + 54
37  libqcocoa.dylib               	0x0000001318e59c05 0x1318e3a000 + 130053
38  com.apple.AppKit              	0x00007fff2e968137 _NSViewDrawRect + 139
39  com.apple.AppKit              	0x00007fff2eaca9a0 -[NSView _drawRect:clip:] + 1103
40  com.apple.AppKit              	0x00007fff2eac8f07 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 2143
[... 1094 lines cut ...]
mapped file                        1.1G      770 
shared memory                      656K       20 
===========                     =======  ======= 
TOTAL                             76.8G     4000 
TOTAL, minus reserved VM space     4.0G     4000 

Model: iMac18,3, BootROM 180.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3,8 GHz, 40 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro580Item, Radeon Pro 580, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334232202020
Memory Module: BANK 0/DIMM1, 16 GB, DDR4 SO-DIMM, 2400 MHz, 0x859B, 0x435431364734534644383234412E4D313646450
Memory Module: BANK 1/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334232202020
Memory Module: BANK 1/DIMM1, 16 GB, DDR4 SO-DIMM, 2400 MHz, 0x859B, 0x435431364734534644383234412E4D313646450
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1440.1)
Bluetooth: Version 7.0.3f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: iPhone, Ethernet, en5
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST2000DM001, 2 TB
USB Device: USB 3.0 Bus
USB Device: BUP BK
USB Device: My Passport 259F
USB Device: External USB 3.0
USB Device: iPhone
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 41.3

Vandrasc, 2020-02-21:
Devs, this is the same as our internal ticket CURA-7202. 

@Gipielle  thank you for your feedback!

Gipielle, 2020-02-21:
Ok perfect !

Vandrasc, 2020-02-21:
The fix will be included in the 4.5 stable release.

Gipielle, 2020-02-21:
When will be released version 4.5 stable ?
Because with Catalina 4.4 don't work anymore

kr15uk, 2020-02-25:
@Gipielle works fine, I'm running 4.4.1 and 4.5.0b side by side on macOS 10.15.3

Ghostkeeper, 2020-02-27:
4.5 stable was released yesterday. Could you try this again?

kr15uk, 2020-02-27:
Brilliant. All fixed now, tested on 3x machines. Thanks for sorting this out.

closed 2020-02-27 by Ghostkeeper