Ultimaker/Cura #7145: 4..4.1 Dual Extrusion Gcode wiith wrong coordinates for the "Move near Prime Tower"

Pomaroli, 2020-02-21:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.4.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windwows 10

**Printer**
(Which printer was selected in Cura?)
Crealitiy CRX Pro V2

**Reproduction steps**
1. Added a Dual extrusion model 
2. Activate Dual Extrusion in the Profil 
3. Set the Tower to X50 and Y50
4. Generate Gcode 
5. Search for the String "Move near prime tower"

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Search for the string "Move near prime tower" in the gcode.
It is always X295 Y265 with my Machine profil CRX. this is wrong and will hit badly the Y End Stop

Code snippet

....
G1 F600 Z1.3
G92 E0

G92 E0 ;reset extruder distance
G1 F800 E-5 ;short retract
**G1 F2400 X295 Y265 ;move near prime tower**
G1 F2000 E-93 ;long retract for filament removal
G92 E0 ;reset extruder distance
G90
T1
G92 E0
.......

**Expected results**
(What should happen after the above steps have been followed.)
Should generate the correct coodinates near tower.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Pomaroli, 2020-02-23:
Issue is related to the Extruder Start/End GCode as the XY coordinates is part of the sequence. Generally i think there have to be more significant changes to be made to cura for Printer wit 2-1 nozzle.

Ghostkeeper, 2020-02-27:
Cura doesn't have a definition for the Creality CR-X Pro V2. Since the issue is with the extruder start and end position that it needs to travel to in order to perform a switch, it needs to be adjusted in the printer definition, but we can't adjust your printer definition since we don't have it. You seem to have modded it into your Cura installation manually.

I think you'll have to take this up with whoever created that printer definition you're using.

closed 2020-02-27 by Ghostkeeper