Ultimaker/Cura #7146: Color generic PLA

kdomkus, 2020-02-21:
When I have a print that uses both print heads, it would help if I could change the colors being shown. If I have generic PLA, the color shown is yellow no matter what. So to tell them apart, I tell it I'm using Ultimaker Green (or whatever) but when it goes to print, I have to tell it to override the material selected. It would be nice if I could choose different colored generic PLA's.

**Is your feature request related to a problem? Please describe.**
Since I picked Ultimaker to get color separation, I have to override material selection when it goes to print.

**Describe the solution you'd like**
I would like to chose a color for my 'generic PLA' other than yellow.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-02-22:
You can duplicate the Generic PLA and give it a different color. Since it is a duplicate it will still have the same identifying code as the original Generic PLA.

kdomkus, 2020-02-24:
Perfect.

Thanks

closed 2020-02-25 by mahtDFR