Ultimaker/Cura #7149: [4.4.1] Pause At Height plugin only works with Layer-No.-Option

Bodengriller, 2020-02-23:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Custom FFF (SWX1)

**Reproduction steps**
If you run the Pause At Height plugin and use "Height" with a height as prefix, no pause command will land in the Gcode!
(I have tried several heights and models, as well as altitude selection like 2 or 2.0 or 2,0)
If you use the option "Layer No." instead, everything works fine.

rburema, 2020-02-25:
This seems to be a duplicate of #7090 
If so, this should be fixed in the upcoming 4.5 -- to confirm, you could try the [4.5-BETA](https://github.com/Ultimaker/Cura/releases/tag/v4.5-beta).

mahtDFR, 2020-02-25:
Closing this off as it's a duplicate of #7090

closed 2020-02-25 by mahtDFR