Ultimaker/Cura #7150: 2-1 Nozzle needs some chagnes to cura

Pomaroli, 2020-02-23:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])
i noticed some issues with cura related to a 2-1 Nozzle printer like Ceeality CRX is very challenging. 
Is there a possibility to have a configuration set for this type of printer ?

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
the Solution should be similar to octoprint were you can set a 2-1 nozzle printer which will be considered while slicing 2 color objects. 

Issues found so far which could be solved if there is a better setup for 2-1 nozzles printers 
- Nozzle End Code with fix tower position has to be parametrable -> see issue 7145
- Temperatur standby temperatur set while changing from one to the other nozzle is wrong as a 2-1 nozzle printer has only one hardware nozzle and the temperature should not be set to standy temperature while printing.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Change the gcode manually to  be usable.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

All printers with 2-1 nozzle

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-02-24:
Cura 4.5 (now in beta) has an option for a shared heater in the Machine Settings. Have you tried that?

Pomaroli, 2020-02-24:
Not yet but i will try out thanks

Pomaroli, 2020-02-24:
Loaded Cura 4.5 Beta and noticed first that still the issue were the Filament is loaded after start line to prime the nozzle.
.......
T0
M82 ;absolute extrusion mode
G21 ;metric values
G28 ;home all
G90 ;absolute positioning
M107 ;start with the fan off
G1 F2400 Z15.0 ;raise the nozzle 15mm
M109 S220.0 ;Set Extruder Temperature and Wait
M190 S50.0; Wait for bed temperature to reach target temp
T0 ;Switch to Extruder 1
G1 F3000 X5 Y10 Z0.2 ;move to prime start position
G92 E0 ;reset extrusion distance
G1 F600 X160 E15 ;prime nozzle in a line
G1 F5000 X180 ;quick wipe
G92 E0 ;reset extrusion distance
G92 E0

T0 ;switch to extruder 1
G92 E0 ;reset extruder distance
G1 F2000 E93 ;load filament    ------>>>>>Filament is loaded to late after first prime 
G92 E0 ;reset extruder distance
M104 S220.0
G92 E0
G1 F2700 E-6.5
;LAYER_COUNT:417
;LAYER:0
.......

fieldOfView, 2020-02-24:
Did you enable the Shared Heater option in Machine Settings?

Pomaroli, 2020-02-24:
yes i did. 
the issue is that the "Start G-Code" is running before the "Extruder Gcode-Start" and the filament is loaded on the "Extruder Gcode-Start". I did not found a solution yet which is working as the "Extruder Gcode-Start"  is running on every Tool change. So if we load the filament on "Start G-Code" it will load twice the amount of filament. The only thing would be if the "Extruder Gcode-Start" could be run before "Start G-code".

Ghostkeeper, 2020-02-28:
I suggest that you add this line of g-code in your starting g-code just before the prime line then:

    G1 F2000 E93 ;load filament

The extruder start g-code should not be run at all before the first layer, but it is. This is considered a bug currently: https://github.com/Ultimaker/Cura/issues/6160

Pomaroli, 2020-02-28:
Make sense to me, ist there a version were #6160 already solved ? otherwise if i insert the line now with the bug it would extrud twice the amount of E93
Thanks

Ghostkeeper, 2020-02-28:
No, we haven't solved that yet :disappointed: Sorry.

Pomaroli, 2020-02-28:
no problem i have a workaround for no, Thanks

Pomaroli, 2020-03-02:
There is another small issue with the retract amount. I noticed when retracting the filament there is always less material as expected. The reason is because a certain amount of filament is fluid in the hot end which will be not retracted. Therfore there is always a gap between the Amount of filament which needs to be refill after e retract. How can this be setup ?

Ghostkeeper, 2020-03-02:
You can try increasing the Retraction Extra Prime Amount setting.

The material that stays inside the nozzle is not a problem. It's going to be pushed out when the filament is unretracted again. The material that oozes out during the travel move is a problem that can be compensated for with Retraction Extra Prime Amount.

Pomaroli, 2020-03-02:
i agree but is there not a gap when the filament is unretracted ?  I tested this and the filaments retraction amount is  E-90 and on its next retraction it will push back E90 which is a gap of more or less the  high of the hot end 10mm. Should it not retract E-90 and unretract E100 ?

smartavionics, 2020-03-02:
Hi @Pomaroli , can you please provide a complete example gcode that shows what you are describing. Thanks.

Pomaroli, 2020-03-02:
Below a Code example made wird Cura 4.5 for a 2 Color Print 

;FLAVOR:Marlin
;TIME:31725
;Filament used: 16.0718m, 3.87348m
;Layer height: 0.15
;MINX:85.625
;MINY:115.625
;MINZ:0.3
;MAXX:170.858
;MAXY:188.818
;MAXZ:63.7
;Generated with Cura_SteamEngine 4.5.0-beta
T0
M82 ;absolute extrusion mode
G21 ;metric values
G28 ;home all
G90 ;absolute positioning
M107 ;start with the fan off
G1 F2400 Z15.0 ;raise the nozzle 15mm
M109 S220.0 ;Set Extruder Temperature and Wait
M190 S50.0; Wait for bed temperature to reach target temp
T0 ;Switch to Extruder 1
G1 F2000 E90 ;load filament
G92 E0 ;reset extrusion distance
G1 F3000 X5 Y10 Z0.2 ;move to prime start position
G1 F600 Y150 E15 ;prime nozzle in a line
G1 F5000 Y180 ;quick wipe
G92 E0 ;reset extrusion distance
G1 F2000 E-90;
G92 E0;
G1 F2400 Z15.0 ;raise the nozzle 15mm
**--------------------------->>>>> Start procedure End**

;Dual Extrusion
T1 ;Switch to Extruder 1
G1 F2000 E90 ;load filament
G92 E0 ;reset extrusion distance
G1 F3000 X10 Y10 Z0.2 ;move to prime start position
G1 F600 Y150 E15 ;prime nozzle in a line
G1 F5000 Y180 ;quick wipe
G92 E0 ;reset extrusion distance
G1 F2000 E-90;
G92 E0;
G1 F2400 Z15.0 ;raise the nozzle 15mm
G92 E0

T0 ;switch to extruder 1
G92 E0 ;reset extruder distance
G1 F2000 E90 ;load filament   **-------->>>>> Load filament of T0 E90**
G92 E0 ;reset extruder distance **----->>>>> Set Zero for E Start printing first layer**
M104 S220.0
G92 E0
G1 F2700 E-6.5
;LAYER_COUNT:417
;LAYER:0
M107
G0 F3600 X94.427 Y119.721 Z0.3
;TYPE:SKIRT
G1 F2700 E0
G1 F1800 X94.926 Y119.386 E0.02999
G1 X95.472 Y119.083 E0.06114
G1 X96.957 Y118.289 E0.14515
G1 X97.607 Y117.953 E0.18166
G1 X98.16 Y117.724 E0.21152
**.....Start printing Layer for T0

.....End Printing Layer fot T0**
G1 F1800 X135.485 Y157.547 E207.53171
G0 F3600 X135.878 Y157.374
M104 S220
G1 F1800 X136.231 Y157.727 E207.55661
G1 F1200 E191.55661
G1 F600 Z1.3
G92 E0

G92 E0 ;reset extruder distance
G1 F800 E-5 ;short retract
;G1 F2400 X295 Y265 ;move near prime tower
G1 F2000 E-90 ;long retract for filament removal  **---->>>> Retract fot T0 E-90**
G92 E0 ;reset extruder distance   **--->>>> Set Zero for T0**
G90
T1
G92 E0

T1 ;switch to extruder 2 **---->>>> Load T1**
G92 E0 ;reset extruder distance
G1 F2000 E90 ;load filament **--->>> Load Filament for T1**
G92 E0 ;reset extruder distance  **--->>>> Set Zero for T1**
M104 S220.0
M105
M109 S220
G1 F2700 E-6.5
;LAYER_COUNT:417
;LAYER:0
M107
G0 F3600 X94.427 Y119.721 Z0.3
;TYPE:SKIRT
G1 F2700 E0
G1 F1800 X94.926 Y119.386 E0.02999
G1 X95.472 Y119.083 E0.06114
**.....Start printing Layer for T1

.....End Printing Layer fot T1**
G0 F5400 X138.735 Y157.683
G1 F1800 X140.039 Y156.379 E83.85648
G0 F5400 X139.401 Y156.451
G1 F1800 X138.51 Y157.342 E83.88792
G0 F5400 X138.305 Y156.981
G1 F1800 X138.738 Y156.549 E83.90317
G1 F1200 E67.90317
G1 F600 Z1.45
G92 E0

G92 E0 ;reset extruder distance
G1 F800 E-5 ;short retract
;G1 F2400 X295 Y265 ;move near prime tower
G1 F2000 E-90 ;long retract for filament removal **---->>>> Retract fot T1 E-90**
G92 E0 ;reset extruder distance **--->>>> Set Zero for T1**
G90
T0
G92 E0

T0 ;switch to extruder 1 -**---->>>>>> Load T0 for 2 Layer**
G92 E0 ;reset extruder distance
G1 F2000 E90 ;load filament **---->>>>> Load Filament E90 <<<<<-------- This should be more then 90 because of the melted Filament in the nozzle (lost printing the Tower) from the last retraction**
G92 E0 ;reset extruder distance  **---->>>>> Set Zero for T0**
M104 S220.0
M105
M109 S220

smartavionics, 2020-03-04:
Don't you just need to set a value for Nozzle Switch Extra Prime Amount ? i.e. 5 to match the last retract before the switch?

Pomaroli, 2020-03-04:
No Luck, i tried to set "nozzle switch extra prime amount" but did not take any effect on the gcode. maybe it is bugy or i did something wrong.

smartavionics, 2020-03-04:
Is it possible for you to supply a project file? Without something to slice and look at the results it's hard to understand the problem. If you can, please do File -> Save and then zip or rename the .3mf file and attach to this issue. Thanks.

Pomaroli, 2020-03-04:
[Untitled.zip](https://github.com/Ultimaker/Cura/files/4289519/Untitled.zip)

here it is

smartavionics, 2020-03-05:
Thanks for the file. I have looked at some gcode produced by that profile and have noticed the following:

1 - you have a nozzle switch retraction distance set to 16 which is probably not wanted because you are doing the 90 mm retract/prime explicitly in the extruder start/stop scripts.

 2 - the un-retract that occurs before the prime tower does include an amount due to nozzle switch extra prime amount so I think that is working.

3 - arguably, the nozzle switch should occur over the prime tower and not over the model but I guess that would need a modification to the engine to achieve.

Pomaroli, 2020-03-05:
Thanks for you Time to investigate. 
1. I did some tsting with retraction distance but it does not take any effect to the Gcode. Maybe if this behavior can be corrected there is no need to do it in the start/stop scripts.
2. This setting was made by me for testing, but it does not take any effect to the gcode.
3. i Agree

Ghostkeeper, 2020-03-06:
I do get differences if I take your 3MF file, load two cubes in with different extruders, and then change the Nozzle Switch Retraction Distance from 16 to 90:

[image]

But yes, your extruder switching g-code already does a 90mm retraction so this will now retract way too much. But as far as I can see, Cura is behaving as it should here.

Pomaroli, 2020-03-06:
Maybe its working for 2-2 Nozzle but not for 2-1. Do you have a Machine with 2-1 nozzle ? 
What i can say is that there was no retraction on the gcode based on the settings for.
Did you try also to merge a model like it is usually for 2 color print ?
See this post

https://ultimaker.com/learn/get-started-with-cura-printing-with-two-colors

smartavionics, 2020-03-06:
I tested using your profile above and I see it had the shared heater option set. As far as I could tell, the retraction was working OK (see my above comments).

Pomaroli, 2020-03-06:
ok I agree i got it,  on my latest test i disabled the Toolchange start/stop gcode, instead i set a retraction of 90 in the Dual Extrud settings. I get the same result as you now.
But what is missing is the load of the filament for after the first toolchange from t0 to t1 and the according Tower prime cycle. Can you please check this ?

G1 X145.134 Y162.105 E227.27413
G1 X146.11 Y161.667 E227.35599
G1 X146.111 Y161.647 E227.35752
G0 F1620 X146.125 Y161.327
M104 S220
M204 S5000
;MESH:Moai_Lines.stl
G0 F3600 X146.183 Y161.385
G1 F1200 E137.35752
G1 F600 Z1.3
G92 E0
T1
G92 E0
M105
M109 S220
G1 F2100 E-5
G0 F3600 X146.183 Y161.385 Z1.3
G1 F600 Z0.3
G0 F3600 X148.892 Y154.947
G0 X153.408 Y151.22
M204 S500
;TYPE:SKIRT
G1 F2100 E0
G1 F1800 X154.352 Y151.722 E0.10002
G1 X154.917 Y152.034 E0.16039

smartavionics, 2020-03-07:
It's no good just posting a fragment of gcode like that. I need to see the whole file (or at least what comes before).

Pomaroli, 2020-03-07:
Here you have the hole file attached, please the the first layer when changing to the second color (tool T1) It does the skirt without loading filament

[CCRX_Moai_Body_Cura_beta.zip](https://github.com/Ultimaker/Cura/files/4301966/CCRX_Moai_Body_Cura_beta.zip)

Ghostkeeper, 2020-03-09:
I can see what you mean, now. It happens with shared heaters or without (as the heater control really has nothing to do with this).

Cura is making the assumption that all extruders' filaments start with the material primed at the nozzle's tip, at E0. So for the first switch to T1, it assumes that this material was at the nozzle tip. It'll retract it 5mm, move to a starting location, unretract 5mm and then start printing. Meanwhile the other material is properly retracted 90mm after it's done its first layer. This will get properly unretracted 90mm when it needs to start again. And for the rest of the print both extruders get retracted/unretracted correctly.

So the real problem is that Cura is not told where the filament is at the beginning of the print. It just assumes it's at position 0, in the nozzle's tip. If you have multiple filaments going into one nozzle this is obviously not a realistic assumption. Cura doesn't have a way to give this initial position currently, so I think the only real solution is to use the start g-code and extruder start/end g-codes at the moment.

From Ultimaker, this will not get any big priority as it's not a concern for our own printers which have the G280 implemented (which primes the nozzle). So I will mark this issue as deferred, to give you more realistic expectations.

Pomaroli, 2020-03-09:
Ok i understand, and what is happening with the Tower Prime at the beginning of T1 which is missing too ? I have to say im not happy with it, as Cura is provîding this type of setup but its clearly a bug which every 3d printer with 2-1 nozzle will have. i my opinion this should be fix as soon as possible,

Ghostkeeper, 2020-03-09:
Instead of a prime tower, Cura will print a ring around the bed adhesion with the other extruder to prime it. This improves bed adhesion.

Pomaroli, 2020-03-09:
If i'm right this was my workaround to setup a skirt with the t1 extruder otherwise it will not prime anything.

Pomaroli, 2023-06-30:
Hi Thanks for Reminder, there are 3 Years between :) You can close this one. THanks