Ultimaker/Cura #7151: 请求给“排队打印”加一个每次打印前停止挤出的选项

DanielStake, 2020-02-24:
在“排队打印”后第二个模型的时候。因为喷嘴在重新升温，出现短暂的停留。这时喷嘴会不断挤出影响下一个模型的打印质量，使用希望可以在“排队打印”里添加一个每次打印前停止挤出的功能。

fieldOfView, 2020-02-24:
Please fill in the feature template, and use english. English isn't my native language either. I - and most of the other developers with me - also put effort into posting and answering questions in a language that is not our native language, so we expect you to at least try to do that too.

DanielStake, 2020-02-24:
**Is your feature request related to a problem? Please describe.**
Excuse me, my question is: while waiting in line to print, the nozzle prints the first model. It will then wait for the heat to finish above the next print model location. At this time, the nozzle will not stop extrusion, which will affect the printing of the next model.

**Describe the solution you'd like**
In response to this problem, I would like to ask you to add a "stop extrusion before printing each model until printing begins" feature in "queued printing". Ensure the print quality of later models.

**Describe alternatives you've considered**
After printing the first model, the nozzle needs to confirm the temperature. Wait until the temperature is set before you continue printing. I had a plan for that. Can you skip the step to confirm the temperature in the middle to reduce the waiting time of the nozzle.

**Affected users and/or printers**
I think all users of Cuar software will benefit from the improvement, because the function of "queuing for printing" can help users to reduce operation time and improve the quality of model printing.

**Additional context**
I'm sorry that I didn't know the rules at the beginning.
I quite understand your request. Since my English is not good, please forgive my choice of words. thank you!

Ghostkeeper, 2020-02-25:
So I think you're saying, it should retract between models while printing in one-at-a-time mode?

fieldOfView, 2020-02-25:
I thought he says that the nozzle rests on the previously printed part while heating up to initial print temperature for the next object. It should probably be raised a little and/or move to a safe position.

gailulun, 2020-02-25:
我想你的意思是不是每次打印完成后喷头温度还没降下来的时候会有料从喷头里流出来。那是因为液态的材料流动性太强，自己流出来的，机器并没有挤出。我想是不是可以每次打印完成后可以”回抽“一下。
//I think you mean that every time the temperature of the nozzle has not dropped after printing, there will be material flowing out of the nozzle. That's because the liquid material is so fluid that it flows out of itself and the machine doesn't extrude it. I wonder if It can "draw back" every time after printing.(machine translation)

DanielStake, 2020-02-26:
"So I think you're saying, it should retract between models while printing in one-at-a-time mode?"
"I think you mean that every time the temperature of the nozzle has not dropped after printing, there will be material flowing out of the nozzle. That's because the liquid material is so fluid that it flows out of itself and the machine doesn't extrude it. I wonder if It can "draw back" every time after printing.(machine translation)"

Yes，you are right! And it will wait for about 5 seconds to confirm the temperature when printing the model one by one.I found a piece of code in one-at-a-time mode:
G0 F3600 X170.067 Y107.774 Z12;Complete model printing and raise to 12mm
G0 X120.474 Y132.724
M140 S50
M105
M190 S50
M140 S50
M105
M190 S50
M104 S180
M105;To obtain the temperature
M109 S180;Set the temperature and wait
;LAYER_COUNT:34
;LAYER:0
M205 X8.18 Y8.18
;MESH:??????.STL
G0 F1800 X130.076 Y138.264 Z0.3;Start printing the next model, the first layer height is 0.3mm
//I tried to delete 
'M105
 M109 S180' and than it hardly has to wait to print the next one.

Ghostkeeper, 2020-02-28:
You didn't copy enough of the g-code to see whether or not it retracts.

When I tried to reproduce this issue, I'm seeing that it does retract the filament and moves away from the print while it's heating up, though. Here is my snippet of g-code. I set the Initial Layer Build Plate Temperature to 66 so that I could see where it's going to be waiting to heat up. The model is 10mm high (just two cubes). Comments added by me:

```
G1 F1800 X54.589 Y110.237 E118.42299 ; <---- second-last skin line
M204 S5000
M205 X30 Y30
G0 F15000 X54.589 Y110.732
M204 S500
M205 X5 Y5
G1 F1800 X54.08 Y111.241 E118.42694 ; <---- Last skin line
;TIME_ELAPSED:744.337710
G1 F1500 E111.92694 ; <---- !! retract !!
M107
G0 F15000 X54.08 Y111.241 Z15 ; <---- !! move 5mm up !!
G0 X186.74 Y42.99 ; <---- !! Move aside to the next cube's location !!
M190 S66 ; <---- Here it's going to be waiting for a while.
M109 S205
;LAYER_COUNT:98
;LAYER:0
; <<continue printing the next model>>
```