Ultimaker/Cura #7155: bad slicing horizontal drill with 0 bottom layer

DorianPIERRE, 2020-02-24:
**Application version**
Cura 4.4.1

**Platform**
Windows 7 64bit

**Printer**
Wanhao Duplicator 9

**Reproduction steps**
1. Part with horizontal drill
2. Ask 0 bottom layer
3. Show slicing detail of the top of drill

**Screenshot(s)**
[image]
[image]
 
**Actual results**
No sliced layer on the top of drill

**Expected results**
Wall layer on the top of drill

**Project file**
[Uploading WD9_TestPourProfilSkate.zip…]()

**Log file**

**Additional information**
I need 0 layer in the bottom of part because i want to flow epoxy resine in printed model for mecanical compression required in case of gyroid infill

I try same on simplify3D with same result...

smartavionics, 2020-02-24:
Hi, the link you give for the project file is broken. My guess is that the model is not valid.

DorianPIERRE, 2020-02-24:
sorry ...
[WD9_TestPourProfilSkate.zip](https://github.com/Ultimaker/Cura/files/4245700/WD9_TestPourProfilSkate.zip)

DorianPIERRE, 2020-02-24:
> Hi, the link you give for the project file is broken. My guess is that the model is not valid.
the model is very simple ... no ?

smartavionics, 2020-02-24:
I recommend two changes to your profile:

1 - turn off Connect Top/Bottom Polygons as that makes the concentric skins ugly.

2 - set Bottom Layers to a value > 0 and Initial Bottom Layers to 0 to get the open bottom.

Hope this helps!

[image]

DorianPIERRE, 2020-02-24:
I try tomorow but it looks good ...
Thanks

DorianPIERRE, 2020-02-25:
Ok for me.

closed 2020-02-25 by DorianPIERRE