Ultimaker/Cura #7157: [4.4.1] Half support infill layer added

Block137, 2020-02-25:
**Application version**
4.4.1 and older

**Reproduction steps**
1. Enable support generation
2. Set support infill layer thickness to double of layer height.

**Screenshot(s)**
Coming soon

**Actual results**
There are tiny support infill dots added at half of Support Infill Layer Thickness which significantly increase print time.

**Expected results**
Support infill is skipped entirely every other layer.

**Project file**
Coming soon

**Additional information**
I have support interface always enabled, maybe part of the cause.

mahtDFR, 2020-02-25:
Thanks for the report (and the model). Seems like a bug, I'll put it on the board. 
As a workaround to print this particular model, it would be much easier to print it from the base upwards! But the bug still stands.

Ghostkeeper, 2020-02-28:
This is working as intended actually. If you specify a higher support layer thickness, it'll combine multiple layers together to form a thicker layer (which is rendered in Layer View on the highest of the layers). This combining only works if there actually are multiple layers to combine. If not, it'll leave the original layer as-is.

As a result, you get thicker layers where the volume of the support allows it, but still high quality layers where the support needs to be smooth to support the curvature of the model nicely. We've found that this improves the overhang quality and makes it easier to remove the model afterwards. So we're not going to remove this feature.

I'll close the issue for now, but of course if you have other findings we can still reconsider.

closed 2020-02-28 by Ghostkeeper

Block137, 2020-02-28:
> but still high quality layers where the support needs to be smooth to support the curvature of the model nicely.

Can this be disabled only when support interface is active instead?

Block137, 2020-03-03:
@Ghostkeeper 
> We've found that this improves the overhang quality and makes it easier to remove the model afterwards.

It should be replaced by Support Interface whenever Support Interface is enabled.