Ultimaker/Cura #7159: Number of active post-processing scripts indicator

Deutherius, 2020-02-25:
**Is your feature request related to a problem? Please describe.**
I run a post-processing script 100% of the time, so the icon (hammer and spanner next to slice/save) that would normally alert me to an active post-processing script is always there. There is currently no indication (that I know of) that would alert me when I forget to turn off some additional post-processing script that I don't need anymore but don't usually use.

Example: I needed to slow down some layers on a certain engineering model for higher accuracy. Used ChangeAtZ to achieve that, which worked flawlessly. Then I forgot to turn ChangeAtZ off. Hammer and spanner icon is always present, so that didn't remind me. Feedrate preview does not show speed after post-processing scripts do their thing, so that also couldn't alert me to my mistake. Next model took unnecessary extra time to print as well as having inconsistent outer wall due to different print speed.

**Describe the solution you'd like**
It would be very nice to have a counter next to the already existing icon that would display the current number of active post-processing scripts, something like this:
[image]
I think this would be a simple and effective extension of the already existing alert for the user. I would for example always expect to see a "1", and seeing anything other than that would remind me to check the post-processing scripts before issuing the print.

**Describe alternatives you've considered**
Alternatively, I could always check the post-processing plugins before saving every print. I could also always check the final gcode by loading it into Cura again. Neither of those will work because that slows down my workflow too much. I'm also way too lazy to do that.

**Affected users and/or printers**
All users that permanently use a post-processing script (e.g. for info on the LCD during print) and occasionally use other post-processing scripts.

**Additional context**
-

mahtDFR, 2020-02-28:
Some additional discussion [here](https://community.ultimaker.com/topic/31375-feature-request-number-of-active-plugins-indicator/?tab=comments#comment-257993)

fieldOfView, 2020-02-28:
I have created a PR implementing this count here: https://github.com/Ultimaker/Cura/pull/7187

PR #7187 (2020-03-02): Show active postprocessing scripts count
This PR adds a "notification" count to the button that indicates there are postprocessing scripts active. It also adds a descriptive tooltip text, showing the name(s) of the active scripts without having to open the PostProcessing plugin dialog to inspect.

[image]

While fixing this, I also fixed some missing tooltips for ActionButtons.

Fixes #7159
closed 2020-03-02 by nallath